betaflight/betaflight-configurator #2101: Android App OSD

SvenHoick, 2020-06-30:
I can’t move the different options around the screen under the OSD tab to set them in place.
I have the Android app installed on my Samsung Note 4. It is connected to my kwad using a Micro USB to Micro USB cable. Everything else seems to be working except for moving things around on my OSD.

WalcoFPV, 2020-07-04:
@SvenHoick Did you try to long press on the element you want to move before moving it ? The apparence of the element will sightly change, signifing you can move it. And the position of your finger define the position of the element, and not the position of the "preview" of the element. It is not as intuitive as on computers and probably need, in the futher, some adjustements.

SvenHoick, 2020-07-09:
@WalcoFPV I did try to long press on the element. I tried doing it quite a few times as I thought maybe my finger wasn’t hitting exactly where it needed to be. Even went as far as to use the Note 4 pen but that didn’t work either.

Asizon, 2020-07-09:
Yes, definitelly the element doesnt  correct match your finger movement

closed 2020-08-15 by stale[bot]