betaflight/betaflight-configurator #2118: Auto Smoothness is hidden when both RC Smoothing Filter Cutoffs are set to Manual.

levelfifty, 2020-07-09:
1. Describe your problem;
When you select manual for Input Filter Cutoff and Derivative Cutoff, it pushes Auto Smoothness off screen and you cannot edit without changing one of the Filters to Auto.

2. Include ways to reproduce the problem;
select manual for Input Filter Cutoff and Derivative Cutoff

3. Provide as much information as possible about your hardware and software, including:
Version 10.7

[Link to SS of all 4 possible combinations of Cutoff select](https://imgur.com/a/a83LCyf).  Any 1 in manual or 0 in manual and its fine, both to manual and it pushes off screen.

Asizon, 2020-07-09:
@levelfifty yes, this is the way to work for this features, if both are set to manual, autosmothness is disabled

McGiverGim, 2020-07-09:
Exactly, what autosmothness does is to modify this two parameters that are auto internally to a different value. If both are manual there is nothing that autosmothness can do.

closed 2020-08-15 by stale[bot]