betaflight/betaflight-configurator #2124: New Android App .apk Webview Issue

dan-youd, 2020-07-14:
The new Android App works flawlessly on the Google devices I've tested it on. However it doesn't work on to different Amazon Kindle Fire's I've tested it on (Fire 7 2015 and Fire 7 2017). Both devices are on their latest firmware.

When loading BF Configurator it gets stuck on a screen where it asks you to download 'Android System Webview' the problem is that it takes you to the Amazon App Store where of course you can't download it.

To try and resolve this I installed Google Play Services and downloaded 'Android System Webview'. When I tried to open the Configurator this time it loaded to a different screen, but got stuck with a white loading screen saying 'Checking webview applications'.

Upon further investigation into this I've realised that the Kindle Fires have their own modified version of 'Android System Webview' simply called 'Amazon Webview'. If I install Googles version they seem to conflict with each other and cause the app to not load fully, but if I remove the Google version the BF Configurator isn't able to detect Amazons version of Webview.

I understand that the main objective of this App was to work on Google devices (which it does very well), but there are a lot of Amazon devices out their that could also utilise this app due to having OTG support and being based on Android. For instance the SpeedyBee app works fine one them.

I might be wrong, but I get the feeling it would be fairly easy to implement the ability for BF Configurator to detect 'Amazon Webview' on startup and use that on the Kindle Devices.

If you need any more clarification on the issue please let me know.

Rejdukien, 2020-07-17:
Similar issue here, OnePlus One running LineageOS 15.1 (Android 8.1 custom rom) that ships with Webview 69.0.3497.109. Fired up logcat, saw this message on the 'Checking webview applications' screen:

`07-17 14:45:14.698 12003 12003 D SystemWebChromeClient: file:///android_asset/www/node_modules/i18next/i18next.min.js: Line 1 : i18next::interpolator: missed to pass in variable app for interpolating Go in the tab of informations about your device in the settings and tap 10 times on the build version number to enable 'Developper options'. After that, search and go in the developper options and select <b>{{app}}</b> as webview provider.
07-17 14:45:14.698 12003 12003 I chromium: [INFO:CONSOLE(1)] "i18next::interpolator: missed to pass in variable app for interpolating Go in the tab of informations about your device in the settings and tap 10 times on the build version number to enable 'Developper options'. After that, search and go in the developper options and select <b>{{app}}</b> as webview provider.", source: file:///android_asset/www/node_modules/i18next/i18next.min.js (1)`

Sure enough, installed the 'Android System Webview' through the Google Playstore, went into the developer options and changed the Webview Implementation to the current 84.0.4147.89 and it works.

dan-youd, 2020-07-18:
unfortunately the Amazon devices don't allow you to change the default WebView app through developer options and even trying to remove it though ADB doesn't work on latest firmware. Apparently if it had worked Googles version would probably work on Betaflight, but break functionality of apps which require the Amazon version.

closed 2020-08-30 by stale[bot]