betaflight/betaflight-configurator #2127: Android App: App glitches and crashes during use.

evildomain, 2020-07-17:
The Android app connects fine using an otg cable on my Google Pixel 4xl.  Once I start making adjustments to my FC(iFlight F7 TwinG V2.1) it starts to glitch(screen flickers, goes black, won't accept touch input, app closes).  I can restart the app and it will do it again.  Eventually it will work and I can actually change some setting but I never know how many times I might have to restart it.

WalcoFPV, 2020-07-17:
@evildomain Does this issue occur only on few tab of the app or on all of them?

evildomain, 2020-07-17:
> @evildomain Does this issue occur only on few tab of the app or on all of them?

I've tried three FC's, iFlight f7 twinG, iflight f4 and a Goku f7.  All seem to act the same way althought the iflight f7 twinG glitches sooner.  It appears this problem only happens when I'm using the PIDS tab.  I tried to recreate the problem in other tabs on all the FC's but it did not happen, so far.