betaflight/betaflight-configurator #2129: 10.7 does not open a window when starting on Debian KDE

0rsch36, 2020-07-18:
Hi all,

starting the 10.7 betaflight-configurator on a Debian KDE machine (Debian testing/bullseye) does not work. The system monitor shows up 2 processes of betaflight-configurator, but no window will appear. Looking for a solution for this issue, I found issue #2028 with the exact same behaviour.

I purged it and manually removed the betaflight folder under .config and re-installed. Still not working.
I tried out 10.7 RC-1: still not working.
I tried out 10.6: it is working flawlessly.

I have the same Debian setup on another machine (old laptop), 10.7 is working there and I don't know why it wont work on my main machine. Let me know what information you need to fix this. Sincerely

Georg

haslinghuis, 2020-07-18:
`sudo mkdir /usr/share/desktop-directories/` did fix this for me.

See https://bugs.archlinux.org/task/33316

0rsch36, 2020-08-27:
Sorry for very late reply: the directory "desktop-directories" exists already, therefor the command wont work.

haslinghuis, 2020-08-27:
Please look at https://github.com/betaflight/betaflight-configurator/issues/2137

closed 2020-10-11 by stale[bot]