betaflight/betaflight-configurator #2131: Connect to FC Fails on Nightly 529

ZzyzxTek, 2020-07-21:
Up through and including nightly build 528, when USB to FC is plugged into a Win7 computer, the comm port is recognized, shown in comm port field, is listed in the pull-down, and it usually auto-connects (or connects with the connect icon if it didn't automatically).

Starting with build 529, when the USB is plugged in, the comm port is no longer shown, nor does it appear in the pull-down list. It does not seem to detect the new device at all, and no longer connects automatically or manually. All of the usual comm port fixes (ImpulseRC Driver Fixer, etc.) all successfully detect the comm port and device, but do not fix connecting in Configurator.

I also tried a complete clean uninstall/reinstall (also manually deleting the AppData folder), starting with the --disable-gpu command line option, and also running Configurator as administrator, none of which helped.

As noted, this is on a Windows 7 machine.

A follow-on message by @mikeller in the betaflight #configurator slack channel suggested that NW.js no longer supports Win7. The change notes for build 529 indicate "bump nwjs to version 0.47.0", so that looks like a strong correlation, although I don't know enough about NW.js to know if comm ports on 0.47.0 specifically would no longer be expected work.

Given the above, perhaps this is not an issue to be fixed, but I wanted to add it so the decision to support or not support Win7 can be made on this issue, or perhaps some workaround can be implemented for the comm ports to continue working.

If build 529 means the end of Win7 support for Configurator, I would suggest making that well known to the community. Despite the EOL of Win7, I suspect I'm not the only one that uses an old Win7 laptop on the bench to configure Betaflight.

McGiverGim, 2020-08-25:
I will mark this as bug. Maybe simply Windows 7 support has been abandoned in NW.js, but if that is true, we must specify the windows versions supported in the README.md file of the Configurator to give this information to the users.

I will wait to see if someone with Windows 7 can get to a fix.

ZzyzxTek, 2020-08-25:
> I will mark this as bug. Maybe simply Windows 7 support has been abandoned in NW.js, but if that is true, we must specify the windows versions supported in the README.md file of the Configurator to give this information to the users.
> 
> I will wait to see if someone with Windows 7 can get to a fix.

I had some good back and forth with Asier Ruiz (Asizon) on the #configurator slack channel (hasn't scrolled off yet as of 8/25/20). He even built some test builds to try and isolate the problem, but we couldn't find a workaround. With NW.js 0.47.0, Configurator seems to no longer be able to enumerate and connect to the FC STM serial ports.

It seems pretty clear it's related to the NW.js 0.47.0 update that was incorporated into Configurator build 529 (all the builds after that won't connect on Win7 for me as well). NW.js seems to have moved on past Win7 compatibility, at least for the type of serial port support needed to connect to the STM drivers under Win7. Everything else seems to still work under Win7, but not being able to detect and connect to the FC is kind of a deal killer.

FWIW, since I opened this issue, I don't think there's any fix to be made, nor could we find a workaround. As mentioned, probably a good idea to let the 4 or 5 people still using Win7 know they can only use build 528 or lower.

mikeller, 2020-10-24:
@McGiverGim: We could probably add something to the installer to detect the Windows version, and if it is Windows 7 then abort with an error message.
Thoughts?

McGiverGim, 2020-10-24:
It will be easy, it has a directive for that: https://jrsoftware.org/ishelp/index.php?topic=setup_minversion

closed 2020-10-27 by mikeller