betaflight/betaflight-configurator #2132: Installer May Not Clean Up AppData Folder

ZzyzxTek, 2020-07-21:
When moving up/down between nightly builds, normally I had just been using Windows Uninstall function from Control Panel -> Programs and Features and then running the official installer, and have not had any issues with that.

When doing some debugging for another issue, after installing build 529, then uninstalling 529 and going back and installing 526, the following (attached) error dialog appeared when starting 526:
[image]

This is easily fixed when uninstalling 529 by also manually deleting the AppData for Configurator in the \Users\<my name>\AppData\Local\betaflight-configurator folder. If 529 is completely uninstalled in this way, the profile error no longer occurs.

I did not see any errors when updating forward from 526 to 529.

It appears there may be a need when installing to check the AppData for incorrect versions? Although uninstalling and reinstalling an older version is admittedly an edge case.

mikeller, 2021-01-08:
> It appears there may be a need when installing to check the AppData for incorrect versions? Although uninstalling and reinstalling an older version is admittedly an edge case.

This would be highly unusual - the internal application data is managed by NW.js, and should not have to be manipulated by external applications. The way this is currently implemented is correct - the application data is updated when moving to a newer version, and ignored when an older version is used with newer application data.

ZzyzxTek, 2021-01-08:
I have not seen anything like this issue for a long time. I would close this since the installation has evolved since last July.

ZzyzxTek, 2021-01-08:
Also, if I'm not mistaken, we determined that because I was using Win7 back then, NW.js support for Win7 had been discontinued as of build 529, so that might also have something to do with it.

mikeller, 2021-01-08:
@ZzyzxTek: If the problem does not exist any more then please close the issue, thanks.

ZzyzxTek, 2021-01-09:
This issue seems related to the NW.js update that occurred in build 529 plus that I was running on Win7. I have not seen similar issues since. Closing this issue.

closed 2021-01-09 by ZzyzxTek