betaflight/betaflight-configurator #2135: Help! "Failed to close serial port" on save and reboot

wmirza89, 2020-07-23:
Hello - I keep getting this error every time I click save and reboot in betaflight:

2020-07-23 @ 03:25:13 -- Device - Rebooting
2020-07-23 @ 03:25:13 -- Unrecoverable failure of serial connection, disconnecting...
2020-07-23 @ 03:25:13 -- Failed to close serial port

I have no idea what the problem is. I tried uninstalling and reinstalling all drivers/firmware/ and betaflight software. Not sure what is causing this issue please help me fix it.

haslinghuis, 2020-07-23:
This is not the correct place to seek support for these kinds of problems. This is the place to report bugs with the firmware. Please try an alternate source like:

Facebook Group: https://www.facebook.com/groups/betaflightgroup/
Betaflight Slack: http://slack.betaflight.com
RCGroups Thread: https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread/

mikeller, 2020-09-01:
@wmirza89: Is there actually anything that is not working, or is it just an error message (which may be caused by the flight controller unexpectedly disconnecting when rebooting)?

westhwerhwerhwerh, 2020-10-10:
Hi @mikeller and @wmirza89 i been having the exact same issue with my IFLIGHT_SUCCEX_E_F4 board :

2020-10-10 @ 10:14:11 -- EEPROM saved
2020-10-10 @ 10:14:11 -- Device - Rebooting
2020-10-10 @ 10:14:11 -- Unrecoverable failure of serial connection, disconnecting...
2020-10-10 @ 10:14:11 -- Failed to close serial port

Have you managed to fix it in any way ,as im not to sure what the bot means when it says its a pull request issue?

haslinghuis, 2020-10-10:
- Are you able to reconnect using Connect button?
- Is auto-connect enabled?

westhwerhwerhwerh, 2020-10-10:
hi @haslinghuis Thanks for the reply
Yes it's quite weird i can easily reconnect after the error,i have tried with and without auto connect enabled and yet no matter what i do it still comes of with the error.

haslinghuis, 2020-10-10:
Guess it's only informational as long settings are saved and you can reconnect without problems there seems to be no issue.
Besides that we dropped chrome support, so I guess implementation changes in future releases. Thanks for reporting.

haslinghuis, 2020-11-18:
Save and reboot triggers a function using `MSP_SET_REBOOT` and should send a console message `Reboot request accepted`.
But this message doesn't fire - and needs an investigation. Occasionally the message comes up while working on serial communication - so far it looks a timing issue. Communication shutdown due to reboot could be a cause to cripple the request.

haslinghuis, 2021-08-01:
@wmirza89 - is still an issue with latest configurator?

wmirza89, 2021-08-01:
@haslinghuis I do not believe so. I have not used the latest configurator / firmware. Even though I got that message it doesn't mean anything and everything worked well.

closed 2021-08-01 by wmirza89