betaflight/betaflight-configurator #2137: configurator not good on linux OS

illmillrig, 2020-07-26:
I'm not sure what it is, but the configurator runs pretty terrible in centos. Any recent version configurator - it doesnt matter which. also happens with multiple flight controllers from different manufacturers. both f4 and f7. Please don't suggest OS or hardware problems ;) all works fine with configurator in windows

I have to make sure that the configurator is opened before I plug in the USB cable to the FC.

Often times, I switch to a different tab, like 'Receiver', but it just hangs, and does nothing. I have to disconnect, close the configurator, unplug the usb cable, reopen the configurator, and plug back in the FC. 9/10 times, it doesn't recognize the FC and won't connect, even if I try to manually connect. From there on, it's just a cat and mouse game of closing the configurator, unplugging the FC, reopening the configurator, and reconnecting the USB to the FC. Eventually it works again, but then shits itself again.

What's going on here? this should work flawless on linux and be buggy in windows :D

haslinghuis, 2020-07-26:
Do you have the modemmanager installed? It runs perfectly fine on Ubuntu 20.04, including flashing.

illmillrig, 2020-07-26:
> Do you have the modemmanager installed? It runs perfectly fine on Ubuntu 20.04, including flashing.

That seems like an odd pkg to be required lol.  But yeah, I have it already.

It works fine once It finally decides to connect, but switching tabs or save/reboot is scary, cause it's pretty much 50/50 it's gonna get hung up.  I wish I could provide more helpful info

haslinghuis, 2020-07-26:
Sorry was not clear, try disable or removing the modemmanager. This is known to interfere with the com ports.

illmillrig, 2020-08-04:
didnt seem to have an affect.  I don't know what do say.  It's terrible in linux.  I have to reboot into windows just to use betaflight.

haslinghuis, 2020-08-04:
Yeah looks like WSL is taking over Linux desktop. With the Windows Terminal Preview everything gets integrated.
I use Ubuntu because I use ROS and other projects which almost mandate you use it. My favorite is Fedora or Arch, but package management of the former starts over with each release having to wait for binaries. No time for all install from source projects and having to manual update. Experienced some reboots using configurator on linux while flashing firmware this week, so there is something not right, but until now it's intermittent.

illmillrig, 2020-08-05:
> 
> 
> Yeah looks like WSL is taking over Linux desktop. With the Windows Terminal Preview everything gets integrated.
> I use Ubuntu because I use ROS and other projects which almost mandate you use it. My favorite is Fedora or Arch, but package management of the former starts over with each release having to wait for binaries. No time for all install from source projects and having to manual update. Experienced some reboots using configurator on linux while flashing firmware this week, so there is something not right, but until now it's intermittent. The reboot happened also on OSX. But also intermittent, using development builds which are not released yet, so I cannot put a finger on it.

yeah I really like fedora too.  I use centos at work, and use a mix of cent and fedora at home along with raspbian/armbian.  never been too big a fan of arch.

anyways, maybe I need to try a differnt usb cable or something. But even in windows sometimes, it seems the configurator gets stuck and won't connect/re-connect.  both linux and windows require me plugging in usb AFTER opening the configurator, and won't ever connect if I reverse the order.

haslinghuis, 2020-08-06:
I had likewise intermittent experience (freezes, not connecting) with a Matek f722 mini - afterwards it turned out that the USB connector had bad contact.
I have been developing on configurator on Ubuntu 20.04 for a few days and had to test, reconnect and flash a lot without any problems on a Matek F411?

illmillrig, 2020-08-06:
just a note that i didn't start experiencing these problems until more recent versions of the configurator.  I don't recall having issues when bf was still at 3.5 (not sure on the equivalent config version back then)

illmillrig, 2020-08-06:
sometimes I get it to connect, then switch tabs and it freezes.  i almost can never get into a cli

haslinghuis, 2020-08-06:
Did you try another FC (and another cable) if this is related?
Also maybe try a nightly build of the configurator to see if there are any improvements.

illmillrig, 2020-08-26:
all the same.  it's always terrible in linux

illmillrig, 2020-08-26:
not too sure why I have no problems with arduino, but disaster with any flight controller.  I don't think it's hardware or even OS related.  something is not right with the configurator.

Asizon, 2020-08-26:
@illmillrig can you provide information about what Betaflight Configurator versions, exatcly builds are you running?

illmillrig, 2020-08-26:
I've been using 10.7.1, but tried 10.4 and had the same problem

illmillrig, 2020-08-26:
once the configurator gets jammed trying to swtich tabs, it's no good any more.  even if I unplug and replug in the FC, the configurator will no longer recognize it and connect.  I have to close betaflight, unplug, open betaflight, plug in, and maybe it connects, but can't switch tabs without rinsing and repeating this.

haslinghuis, 2020-08-26:
Maybe something got corrupted in the configuration.
Please remove the configurator and manual remove any remainders and reinstall like:
```
sudo dpkg -P betaflight-configurator
sudo rm -rf /opt/betaflight/betaflight-configurator
rm -rf ~/.cache/betaflight-configurator
rm -rf ~/.config/betaflight-configurator
sudo dpkg -i ~/tmp/betaflight-configurator_10.7.0_amd64.deb
```

As you are using a mix of computers, which particular OS / Desktop Environment version are we investigating here?

illmillrig, 2020-08-30:
any hardware, both centos and fedora. multiple usb cables.

haslinghuis, 2020-08-30:
Installed Fedora 32 on Lenovo laptop and using this commands I do no experience any issues yet.
```
$ sudo dnf check-update
$ sudo dnf install libatomic
$ sudo dnf remove ModemManager
$ wget https://github.com/betaflight/betaflight-configurator/releases/download/10.7.0/betaflight-configurator-10.7.0-1.x86_64.rpm
$ sudo rpm -i Downloads/betaflight-configurator-10.7.0-1.x86_64.rpm
```

But need more configuration. Not able to flash firmware because of missing udev rules because I cannot find the right instructions for it, fwupd package should address this. Maybe you have a hint?

ceva84, 2020-09-02:
I have issues with ubuntu 20.04 as wel. 
The proces for betaflight configurator opens, but doesnt show a GUI.
Strangely enough my issues started when i upgraded to 20.04 from 19.10
It ran spotless in 19.10
I tried removing, purging, removing using the instructions mentioned above. Nothing seems to do the trick for now ...

haslinghuis, 2020-09-02:
@ceva84 `sudo mkdir /usr/share/desktop-directories/` fixed this for me after upgrading. With a fresh install there was no need to do this.

ceva84, 2020-09-02:
> @ceva84 `sudo mkdir /usr/share/desktop-directories/` fixed this for me after upgrading. With a fresh install there was no need to do this.

Seems to be a negative for me :( file already exists ...

haslinghuis, 2020-09-02:
On fedora I had to install `libatomic`, but on ubuntu `libatomic1` is already installed. I am using the internal Intel GPU, do you have  a dedicated GPU? (Also users installing a fresh install didn't have problems afaik)

ceva84, 2020-09-03:
> On fedora I had to install `libatomic`, but on ubuntu `libatomic1` is already installed. I am using the internal Intel GPU, do you have  a dedicated GPU? (Also users installing a fresh install didn't have problems afaik)

Internal intel gpu as well here. I will retry some things this afternoon. Keep you posted

McGiverGim, 2020-09-03:
I don't use habitually Linux, son I can't help here too much.

If you find "what changes are needed at OS level", the RPM and DEB packages execute an script before and after installing it, so it can be changed if you find something that is needed.

It is here for RPM:
https://github.com/betaflight/betaflight-configurator/blob/8032b1f4ee9adee23ce728ca00710628661e13a2/gulpfile.js#L644-L645

And here for DEB:
https://github.com/betaflight/betaflight-configurator/blob/8032b1f4ee9adee23ce728ca00710628661e13a2/gulpfile.js#L606-L607

illmillrig, 2020-09-28:
any update on this?  It's really frustrating cause it almost works.  enough to keep trying, but not enough to get anything done, and overall just is really frustrating that it can be so bad on linux, but work fine in windows.

illmillrig, 2020-09-28:
it's really not good needing to unplug/plug in the usb so many times.

illmillrig, 2020-09-28:
`[13300:13300:0928/002927.618181:ERROR:component_loader.cc(169)] Failed to parse extension manifest.`

illmillrig, 2020-09-28:
`[13300:13307:0928/003051.958971:ERROR:serial_io_handler.cc(145)] Failed to open serial port: FILE_ERROR_IN_USE`

illmillrig, 2020-09-28:
if I can even get it to load properly, and change a setting.  clicking 'save and reboot' does nothing. it just sits there.  I unplug, and plug back in, and the setting has reverted.  can't get anywhere in linux unfortunately.

haslinghuis, 2020-09-28:
Try put this rules in and reinitialize:

**udev rules**
```
   $ cat /etc/udev/rules.d/45-stdfu-permissions.rules
   # Notify ModemManager this device should be ignored
   ACTION!="add|change|move", GOTO="mm_usb_device_blacklist_end"
   SUBSYSTEM!="usb", GOTO="mm_usb_device_blacklist_end"
   ENV{DEVTYPE}!="usb_device", GOTO="mm_usb_device_blacklist_end"
   
   ATTRS{idVendor}=="0483", ATTRS{idProduct}=="df11", ENV{ID_MM_DEVICE_IGNORE}="1"
   
   LABEL="mm_usb_device_blacklist_end"
   
   #STM32 DFU Access
   SUBSYSTEM=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="df11", TAG+="uaccess"
```

haslinghuis, 2020-09-28:
Using Fedora for a month now and have used configurator extensive without the problems you describe. Please create the file with udev rules below and reboot:

**cat of udev rules**
```
   $ cat /etc/udev/rules.d/45-stdfu-permissions.rules
   # Notify ModemManager this device should be ignored
   ACTION!="add|change|move", GOTO="mm_usb_device_blacklist_end"
   SUBSYSTEM!="usb", GOTO="mm_usb_device_blacklist_end"
   ENV{DEVTYPE}!="usb_device", GOTO="mm_usb_device_blacklist_end"
   
   ATTRS{idVendor}=="0483", ATTRS{idProduct}=="df11", ENV{ID_MM_DEVICE_IGNORE}="1"
   
   LABEL="mm_usb_device_blacklist_end"
   
   #STM32 DFU Access
   SUBSYSTEM=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="df11", TAG+="uaccess"
```
Not to forget also add yourself to the `dialout` group and `dnf remove ModemManager` if you hadn't done that already (login again)

Could it be there is another serial device trying to hijack the port? Please try with the least amount of serial devices attached to rule this out.

Also found this [NW.JS issue](https://github.com/nwjs/nw.js/issues/7575) regarding this error:
```
[13300:13300:0928/002927.618181:ERROR:component_loader.cc(169)] Failed to parse extension manifest.
```

MotoLab, 2020-09-28:
I'm seeing similar issues on kubuntu 20.04 and then lubuntu 20.04 on an older laptop with 4G RAM. Configurator starts ok, but at some point becomes very laggy. Sometimes. Then with the application closed, the linux system also lags, even though top shows low cpu usage and free shows about 500M RAM used and 1.5G free. Subsequent linux shutdown hangs up and doesn't complete. After linux restart configurator again works normally for some time. Linux doesn't show any issues without running configurator. After restart free shows 348M RAM used.

ModemManager is a lubuntu desktop dependency, but is disabled with 'systemctl disable ModemManager.service'. lubuntu also requires manual package installation for libgconf-2-4 and libatomic1.

cooki3monst3r77, 2020-10-07:
I also have no GUI, but betaflight is running.... Using Ubuntu 20.04. Is there a way to manually launch the GUI?

haslinghuis, 2020-10-07:
To start Betaflight configurator manually using Linux using terminal:
```
cd /opt/betaflight/betaflight-configurator
./betaflight-configurator
```

cooki3monst3r77, 2020-10-07:
> To start Betaflight configurator manually using Linux using terminal:
> 
> ```
> cd /opt/betaflight/betaflight-configurator
> ./betaflight-configurator
> ```

Using this command, I get:

[3972:3972:1007/115902.748234:ERROR:sandbox_linux.cc(376)] InitializeSandbox() called with multiple threads in process gpu-process.
[3935:3935:1007/115902.887198:ERROR:component_loader.cc(169)] Failed to parse extension manifest.
[4000:4000:1007/115903.395673:FATAL:nw_content_renderer_hooks.cc(53)] Failed to load node library (error: libatomic.so.1: cannot open shared object file: No such file or directory)
#0 0x7fe59b710ed9 <unknown>

I can see that the file is in the desired location.......

haslinghuis, 2020-10-07:
Did you install libatomic / libatomic1 (not behind the terminal now)?

cooki3monst3r77, 2020-10-07:
> Did you install libatomic / libatomic1 (not behind the terminal now)?

I installed libatomic-ops-dev. Wrong thing? Sorry, new to setting up betaflight on Linux.

haslinghuis, 2020-10-07:
On Ubuntu please remove that one and install `libatomic1`

cooki3monst3r77, 2020-10-08:
That did the trick! Thank you so very much for the help. I truly appreciate it!

closed 2021-01-08 by github-actions[bot]

illmillrig, 2021-10-03:
same problem with fedora 34 :(