betaflight/betaflight-configurator #2141: Linux, Open serial port for flashing (DFU)

fiechr, 2020-07-29:
Every time I want to flash firmware using the Configurator, I run into this issue:
I have to press _two_ times on "Flash firmware" to actually flash the firmware.
The first click only puts the device in DFU mode (`Device - Rebooting to ROM BOOTLOADER`, "Exit DFU mode" button becomes active).... but then, I get an error message (`Failed to open serial port`). But ignoring that and simply pressing the button again, does flash the firmware successfully. Sometimes the Configurator crashes, before I can press the button again. But then I just need to restart it (DFU mode is still active) and start the flashing process.

This is true for every single device I have flashed (F411, F405 and F722 FCs) on this system.

Maybe this is some timing issue, maybe the serial port isn't ready or fast enough or something like that. It also may depend on the hardware used.

"No reboot sequence" and "Full chip erase" is disabled. No custom baudrate is selected.

Happy to provide more info! I did not find any obvious error messages outside the Configurator.

Connecting the FC:
```
[18797.664511] usb 8-3: new full-speed USB device number 11 using ohci-pci
[18797.857024] usb 8-3: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 2.00
[18797.857030] usb 8-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[18797.857034] usb 8-3: Product: Betaflight STM32F411
[18797.857036] usb 8-3: Manufacturer: Betaflight
[18797.857039] usb 8-3: SerialNumber: 0x8000000
[18797.859149] cdc_acm 8-3:1.0: ttyACM0: USB ACM device
```

Start flashing (Open serial port error message is displayed):
```
[18880.998963] usb 8-3: new full-speed USB device number 12 using ohci-pci
[18881.184797] usb 8-3: New USB device found, idVendor=0483, idProduct=df11, bcdDevice=22.00
[18881.184803] usb 8-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[18881.184807] usb 8-3: Product: STM32  BOOTLOADER
[18881.184809] usb 8-3: Manufacturer: STMicroelectronics
[18881.184812] usb 8-3: SerialNumber: 356637553xxx
```

```
Bus 008 Device 010: ID 0483:5740 STMicroelectronics Virtual COM Port
Device Descriptor:
bLength                18
bDescriptorType         1
bcdUSB               2.00
bDeviceClass            2 Communications
bDeviceSubClass         2 Abstract (modem)
bDeviceProtocol         0 
bMaxPacketSize0        64
idVendor           0x0483 STMicroelectronics
idProduct          0x5740 Virtual COM Port
bcdDevice            2.00
iManufacturer           1 Betaflight
iProduct                2 Betaflight STM32F411
iSerial                 3 0x8000000
bNumConfigurations      1
Configuration Descriptor:
bLength                 9
bDescriptorType         2
wTotalLength       0x0043
bNumInterfaces          2
bConfigurationValue     1
iConfiguration          0 
bmAttributes         0xc0
Self Powered
MaxPower              100mA
Interface Descriptor:
bLength                 9
bDescriptorType         4
bInterfaceNumber        0
bAlternateSetting       0
bNumEndpoints           1
bInterfaceClass         2 Communications
bInterfaceSubClass      2 Abstract (modem)
bInterfaceProtocol      1 AT-commands (v.25ter)
iInterface              0 
CDC Header:
bcdCDC               1.10
CDC Call Management:
bmCapabilities       0x00
bDataInterface          1
CDC ACM:
bmCapabilities       0x02
line coding and serial state
CDC Union:
bMasterInterface        0
bSlaveInterface         1 
Endpoint Descriptor:
bLength                 7
bDescriptorType         5
bEndpointAddress     0x82  EP 2 IN
bmAttributes            3
Transfer Type            Interrupt
Synch Type               None
Usage Type               Data
wMaxPacketSize     0x0008  1x 8 bytes
bInterval             255
Interface Descriptor:
bLength                 9
bDescriptorType         4
bInterfaceNumber        1
bAlternateSetting       0
bNumEndpoints           2
bInterfaceClass        10 CDC Data
bInterfaceSubClass      0 
bInterfaceProtocol      0 
iInterface              0 
Endpoint Descriptor:
bLength                 7
bDescriptorType         5
bEndpointAddress     0x01  EP 1 OUT
bmAttributes            2
Transfer Type            Bulk
Synch Type               None
Usage Type               Data
wMaxPacketSize     0x0040  1x 64 bytes
bInterval               0
Endpoint Descriptor:
bLength                 7
bDescriptorType         5
bEndpointAddress     0x81  EP 1 IN
bmAttributes            2
Transfer Type            Bulk
Synch Type               None
Usage Type               Data
wMaxPacketSize     0x0040  1x 64 bytes
bInterval               0
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
(Bus Powered)
```

haslinghuis, 2020-07-29:
Which Linux distro and version are you using? There seems to be a driver problem? I updated the doc for Building in Ubuntu 20,04 and do not experience this issue?

fiechr, 2020-07-29:
I'm running Manjaro, which is Arch-based, Kernel: 5.7.9-1-MANJARO

I don't know about Ubuntu, but I can check my laptop which is using the same system but different Hardware (Intel-based instead of AMD-based).

fiechr, 2020-07-30:
I've tested this on another system with completely different hardware, but same OS and got the same issue.

[image]

Is there some kind of debug mode or verbose log output, to enable for the Configurator?

McGiverGim, 2020-07-30:
Yes if you are able to build the code for yourself.

closed 2020-09-06 by stale[bot]