betaflight/betaflight-configurator #2142: Updated BTF Configurater from 10.6.0 to 10.7.0 now configurator does not recognise the flight controller when the USB is connected and I cannot select any actions tabs including the CLI tab

d605, 2020-07-31:
I have updated the Betaflight configurator from 10.6.0 which was all working fine to the latest 10.7.0 in order to update the flight controller to the latest BTF 4.2 firmware. When plugging in the USB cable and connect in the top left corner I get the configurator version as 10.7.0, the firmware version as 4.1.2 (as not up dated yet) and my target. I dont get the 3D model of the quad on the screen all there is is a purple font box that says BACK in capital letters. I cannot enter into any of the set up tabs including the CLI tab so i cannot enter into boot loader or flash new firmware, it is like the configurator is not recognising the flight controller. I already have the STM32 drivers installed from the previous version and the version appear to be the same so I have not downloaded and installed those at this point. I dont know what the issue is or if there is there a fix for this

d605, 2020-07-31:
I am very sorry, I thank all that has tried to help and thank you Asizon but unfortunately I do not follow. I unable to find any where a short cut or any where to find create a short cut nor can I find an executable icon? I have a folder set up for Betaflight 10.7.0, in here I saved the installer, I ran the installer to also save in this folder. If I now right click on my saved betaflight configurator application and then select properties I am presented with the configurator properties page I posted earlier. I cannot find any executable icons or short cut options. I have been going round in circles for hours

haslinghuis, 2020-07-31:
@d605 Please open Windows Explorer. Click on H: drive. Open (double click) on Betaflight to open this folder. 
You now see the Betaflight Configurator.exe file.
On this icon click right mouse and choose `Send to >` and than choose `Desktop (create shortcut).`
Now on your desktop you find the icon where you add the instructions we provided earlier.

d605, 2020-08-01:
Solved 
Brilliant Haslinghuis thank you for the directions on that I was now able to find the shortcut and apply the change to the shortcut target. 
I then opened the configurator directly from the shortcut icon and connected my quad and all appears to be working fine.
It still doesn't work through the original 10.7.0 configuration download but it does every time through the short cut so I can just use that.
Thank you to all for your efforts and apologies if I wasted any ones time, I appreciate you are all very busy and grateful of your work

mikeller, 2020-08-01:
Dupliceate of #2063.

closed 2020-08-01 by mikeller