betaflight/betaflight-configurator #2143: Incorrect error message when activating mass storage device mode.

hydra, 2020-07-31:
When clicking 'Active mass storage device mode' on an FC which has FIRMWARE that doesn't support MSC the error message appears as below:

[image]

"This operation is not supported by your hardware".

This statement is incorrect, since it's the FIRMWARE that does not support it.

Please change the message to read:

"This operation is not supported by the firmware you are using".

e.g. for a newer MCU for which MSC support has been added the hardware technically probably supports it, just that it's not been written yet.  For an older FC the suggested message is still correct, since the firmware may not/cannot support it.  Either way it's a software thing.

If, the hardware PHYSICALLY cannot support it, then suggest adding another flag and use appropriate messages based on the PHYSICAL and FIRMWARE support status.

hydra, 2020-07-31:
Or, it might also be better not to show the button to activate MSC at all, if the firmware and/or hardware doesn't support MSC.

hydra, 2020-08-31:
@McGiverGim any thoughts?

McGiverGim, 2020-08-31:
In my opinion the cleanest way is to remove the option if not supported. But I don't see this as urgent at this moment. 

This message was added by @mikeller I was waiting to see his opinion about this before try to do any change.

mikeller, 2020-08-31:
@hydra: Please stop spamming for your 'bug reports' if they do not get any reaction. You are already not in good standing with the Betaflight project for copying without proper attribution, any more infringements of the community standards will get you banned.

closed 2020-08-31 by mikeller

hydra, 2020-09-01:
@mikeller 
a) wasn't spamming, was simply bringing this to the attention of the main configurator developer to get some feedback before the stalebot closed the issue.
b) if you feel there's some attribution that needs to be added to some project somewhere please let me know where, you've got my email.  I'm a proponent of giving credit where credit is due, you know that.  As does timecop and ctn.

mikeller, 2020-09-01:
@hydra: If nobody reacts to 'issues' you raise nobody is interested in them, and 'trying to bring to attention' is spamming.
I don't think that it is up to me or anybody to bring to your attention that copying entire releases and claiming any changes in them as your own is against the rules of this repository - it is up to you to educate yourself and follow the rules.