betaflight/betaflight-configurator #2145: Out of memory 'Will not fit in region' compiling STM32F7X2 on stock 4.3

EjectSteve, 2020-08-04:
I tried to compile TARGET=STM32F7X2 but looks like some bit of it was too large? Is there something extra that has to be configured or removed for STM32F7X2 to work? It doesn't even work on stock, unedited code. STM32F411 and MATEKF722SE compiled with no issues.

Linking STM32F7X2
Memory region Used Size Region Size %age Used
ITCM_RAM: 15648 B 16 KB 95.51%
ITCM_FLASH: 0 GB 16 KB 0.00%
ITCM_FLASH_CONFIG: 0 GB 16 KB 0.00%
ITCM_FLASH1: 0 GB 480 KB 0.00%
AXIM_FLASH: 10154 B 10 KB 99.16%
AXIM_FLASH_CUSTOM_DEFAULTS: 8 B 6 KB 0.13%
AXIM_FLASH_CONFIG: 0 GB 16 KB 0.00%
AXIM_FLASH1: 506400 B 480 KB **103.03%**
AXIM_FLASH_CUSTOM_DEFAULTS_EXTENDED: 0 GB 0 GB -nan%
DTCM_RAM: 29708 B 64 KB 45.33%
SRAM1: 76520 B 176 KB 42.46%
SRAM2: 0 GB 16 KB 0.00%
MEMORY_B1: 0 GB 0 GB -nan%
/usr/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld: obj/main/betaflight_STM32F7X2.elf section `.tcm_code' will not fit in region `AXIM_FLASH1'
/usr/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld: region `AXIM_FLASH1' overflowed by 14880 bytescollect2: error: ld returned 1 exit status
Makefile:397: recipe for target 'obj/main/betaflight_STM32F7X2.elf' failed

haslinghuis, 2020-08-04:
I guess you are using the older arm SDK. Are you using WSL or linux? Or Docker?
Please upgrade using `make arm_sdk_install`

McGiverGim, 2020-08-04:
And I think this is the bad repo for the issue, this is for the Configurator, and your problem is with the firmware.

EjectSteve, 2020-08-04:
I'm using docker on windows, I'll have to check how to upgrade arm_sdk_install on it.

haslinghuis, 2020-08-04:
Docker is not really maintained since WSL(2) and upgrading arm SDK will solve it.

Please close this issue as this is for the configurator, not the firmware.

closed 2020-08-04 by EjectSteve