betaflight/betaflight-configurator #2147: Betaflight Receiver Channel Bars not working

Sennadachi, 2020-08-07:
My Problem: I have correctly configured my flight controller according to my hardware. When checking that betaflight recognizes my transmitter inputs via the receivers tab, I get nothing.

The hardware I am using: 

Brand New Mamba F405 MK2 flight controller
Flysky FS-I6 transmitter (with firmware upgrade)
Fli14+ receiver.
AKK Infinite VTX
Foxeer Arrow mini camera

Steps I have done
I flashed my Flight Controller to version 4.2
Enabled serial RX to UART 6 (receiver wired to RX6)
Changed receiver protocol to Serial Based and IBUS
Bound the receiver to the transmitter.
The receiver recognizes the transmitter and rx LED goes solid when tx is on and flashes when tx is off.
Soldering joints are fine.
I have tried a different receiver (the one that comes with the FS-I6 (the big boy rx))
I have tried all of the uarts.
(additional note, the drone used to work flawlessly and then it suddenly stopped)

here is my diff all

[[[[[[[[[

# 
# Building AutoComplete Cache ... Done!
# 
# diff all

# version
# Betaflight / STM32F405 (S405) 4.2.1 Jul 19 2020 / 06:18:01 (caa0d683c) MSP API: 1.43
# config: manufacturer_id: DIAT, board_name: FURYF4OSD, version: 3a35e73b, date: 2019-09-30T05:46:12Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name FURYF4OSD
manufacturer_id DIAT
mcu_id 003700403152501120333647
signature 

# serial
serial 5 64 115200 57600 0 115200

# master
set acc_calibration = -57,-39,-61,1
set serialrx_provider = IBUS
set blackbox_p_ratio = 0
set blackbox_device = NONE
set dshot_bidir = ON
set gyro_1_align_yaw = 1800

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2
[... 1104 lines cut ...]
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 65
set tpa_breakpoint = 1350
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end
]]]]]]]]]

thank you for any responses or help
Sennadachi

McGiverGim, 2020-08-25:
I'm not too sure about what the problem is, but:
1. This seems more a user problem. Github is for bugs or development issues. 
2. If you think is a development problem, use the firmware Github https://github.com/betaflight/betaflight/issues, not the Configurator one. If the FC can't read the values of the TX is clear that is more a firmware problem and not a Configurator problem.
3. Please, when attaching information, don't attach the `dump` command, usually the `diff` command is enough.

I close this.

closed 2020-08-25 by McGiverGim

fusionfpv, 2020-09-08:
HOLA YO TENGO EL MISMO PROBLEMA SI LO SOLUCIONASTES PODRIAS DECIRME COMO LO AS ECHO GRACIAS