betaflight/betaflight-configurator #2148: Virtual Flight controller for testing and tech support

rhys100, 2020-08-08:
Moved from https://github.com/betaflight/betaflight/issues/10085 as I think this is more a configurator issue.

**Is your feature request related to a problem?**
I work at a store and we get calls every day about flight controller settings, how to set up, binding etc. We often have to find a quad so we can plug in and connect to betaflight to walk them through the menus and settings.

**Describe the solution you'd like**
I would like to be able to select Virtual Flight Controller, hit connect and it loads a virtual fc so I can walk through the menus of betaflight as I am helping customer.

**A suggested interface would be a dropdown option like this**
[image]

I note that SITL is a possible current workaround however it only works on Linux and is not user friendly.  - https://github.com/betaflight/betaflight/blob/master/src/main/target/SITL/README.md?fbclid=IwAR05PE9_Q1gt-qz0mA-lWBAWOukv11UnLgqiPH0syAo8Sk4DBIFboRlcpzE

rhys100, 2020-08-08:
I have put my money where my mouth is and kicked this off with a $150 USD bounty reward https://www.bountysource.com/issues/92256565-virtual-flight-controller-part-1-load-default-virtual-for-testing-tech-support You may back this as well if you think it's helpful.

McGiverGim, 2020-08-08:
Only as a comment: in windows you can start the sitl target using WSL and connect with it. But yes, is not user friendly.

visdauas, 2020-09-29:
Hi, I started working on this. Do you want the whole thing simulated like in https://github.com/betaflight/betaflight/blob/master/src/main/target/SITL/README.md (Might be hard to do on Windows/Android) or is it enough for me to add a Virtual option to the ports, disable msp and fill the GUI with some dummy config data?

rhys100, 2020-09-30:
The latter in my opinion would be fine, as long as it can link in with part 2 at a later stage https://github.com/betaflight/betaflight-configurator/issues/2149

visdauas, 2020-09-30:
Alright, If under part 2 you mean you want to use these https://github.com/betaflight/unified-targets/tree/master/configs/default as default presets then, I can make the 2 parts with this approach.

rhys100, 2020-09-30:
perfect. @mikeller you happy with this?

McGiverGim, 2020-09-30:
The "easy" part is the first, is "only" to put some default values in the FC object (that is what the MSP protocol call does) and "block" any read/write call to the MSP protocol... With this all the tabs will show this default values. Maybe with this is enough for the purpose. And I think we can accept this kind of PR (a lot of people is interested in loading the tabs without a FC) if it does not modify too much code that can lead to a lot of maintenance in the future. I'm not too sure how to maintain this "default" values while the MSP protocol grows in an easy manner.

The second part is more difficult. The files you are talking (unified-targets or diffs) are at a CLI format, that is very different that the MSP values that we have. You will need to "translate" from one to the other, and this can change in the time (we try to maintain it, but sometimes we rename some CLI element or update some MSP command).

mikeller, 2020-10-01:
Not sure how much value there is in this 'simple' approach. In my opinion it is too simple, as it will not give an accurate representation of how the configurator looks or behaves when connected to an actual flight controller. The reason for this is that a lot of the UI is driven by a given flight controller's hardware configuration (e.g. showing the OSD tab or not), and by the firmware version running on the flight controller (what features are shown). I am not sure if having a configurator UI that does not reflect what the user actually sees will be much value to help a user over the phone. What are your thoughts on this @rhys100?

visdauas, 2020-10-01:
I can add an option to select the firmware version before you connect, and simply unlock everything that's hardware dependent. I can also add a menu where the users can select how many servos and other hardware they would like to connect. I believe this way users would be able to at least look at the features available with their specific hardware and firmware version in mind. I know this is not a full solution but it's still a lot better than nothing. This approach still has a smaller footprint on the codebase compared to SITL or other firmware simulation/emulation methods, but I'm open to suggestions @mikeller.

rhys100, 2020-10-01:
I think the simple approach would be sufficient for 80% of cases as I only need to see the interface and where sections are located to guide the user - 1. Tell them what tab its on, 2. Whereabouts and what it is called on the actual tab.

For advanced support you could then load up a unified target and look at the default resources and settings and see if they have mucked with any etc such as changing ports or resources that they should not of done.

visdauas, 2020-10-03:
Alright, just to confirm if everybody is okay with what we discussed.
- You will be able to select the firmware version before you connect.
- The default config will unlock every feature available with the selected firmware version.
- I will only modify the initialize function in the pages that require more than modifying the fc config, this will make this feature easy to maintain.
Can I continue working on this @mikeller @rhys100 @McGiverGim?

mikeller, 2020-10-04:
@visdauas: Sounds ok to me.

neildotwilliams, 2020-10-28:
I logged in looking for this exact feature in the configurator, so I'm grateful for people taking a look at this.

Not wanting to add scope creep but could it be considered to add the ability to import a cli dump from a real FC into the virtual FC?  This might help with tech support issues and getting a 'real' view of someones FC. 
 Couldnt this be handy for more than just tech support though, devs, social media influencers, etc

rhys100, 2020-10-29:
Hows it going @visdauas  ?

visdauas, 2020-10-29:
Sorry about the delay, I'm having a little trouble with making every tab useable especially the OSD, but I'm making good progress and it should be ready soon. Also, I was wondering if exporting/importing the whole FC class to JSON would be a better option than parsing the CLI format.

mikeller, 2020-10-30:
I think parsing the CLI format is a very tough ask - there is a lot of 'business logic' between the CLI and what is sent to configurator on MSP. Persisting the FC class is essentially what the backup / restore tab already does, so you could potentially re-use some of that.

visdauas, 2020-11-01:
Thanks for the great idea @mikeller. Can somebody share their JSON backup (possibly with servos, leds, and as many extra things connected as possible) for testing?

mikeller, 2020-11-03:
@visdauas: You can create one yourself, on the 'Setup' tab: https://github.com/betaflight/betaflight/pull/7264#issuecomment-713599897

neildotwilliams, 2020-11-08:
@visdauas Just for a bit of variation, I have attached a JSON config.  Not sure if it flies yet but don't think that matters for you ;-)
Matek F405-AIO F4
Runcam Split (Original)
TBS Unify Pro 5G8 HV SMA
FRSky X4r-SB
[BTFL_backup_wippy_20201108_174026.json.txt](https://github.com/betaflight/betaflight-configurator/files/5506861/BTFL_backup_wippy_20201108_174026.json.txt)

PR #2283 (2020-12-31): Added Virtual Mode (Virtual Fc Part 1 & 2)
Fixes #2148, Fixes #2149. And I believe also helps with this #2259.
- You can now connect in virtual mode and select a firmware API version. This loads a default virtual config that unlocks almost all of the features, the only missing feature is the race transponder because I didn't have the hardware and couldn't find enough information about how it works.
- I modified the OSD so it doesn't require communication with hardware in virtual mode to update the screen.
- The restore feature works, and with this, you can potentially debug other people's conf [... cut]
closed 2020-12-31 by mikeller