betaflight/betaflight-configurator #2150: Filter Targets with text search in Firmware Flasher

rhys100, 2020-08-08:
My suggested feature here is to enable search in the betaflight targets as the list gets longer and longer so it's quicker and easier to find your betaflight target to flash. A suggested GUI interface example below, where you type and it filters the text you have written.

[image]

rhys100, 2020-08-08:
I have put my money where my mouth is and kicked this off with a $15USD bounty reward https://www.bountysource.com/issues/92256568-filter-targets-with-text-search-in-firmware-flasher / You may back this as well if you think it's helpful.

McGiverGim, 2020-08-08:
It's a good idea.

pevecyan, 2020-08-09:
Hi, 
I added this option under pull request #2152 
If this will be accepted please consider giving that money directly to betaflight via donation :)

McGiverGim, 2020-08-25:
@rhys100 this feature request has been developed by @pevecyan and merged.

You can test the nightly build if you want: https://github.com/betaflight/betaflight-configurator-nightlies/releases

rhys100, 2020-08-25:
SO good.! 👍 
[image]

closed 2020-08-25 by rhys100