betaflight/betaflight-configurator #2151: Incorrect flash size shown when flashing EXST targets.

hydra, 2020-08-09:
When connecting an EXST target in firmware upgrade mode and flashing it and the following log messages are generated:

```
2020-08-09 @ 13:47:34 -- Detected device with total flash size 131072 KiB
2020-08-09 @ 13:47:37 -- Erased 131072 kB of flash successfully
```

This is incorrect, the system is reporting the MCU's flash size, and not the size of the EXST Firmware partition being erased and flashed, which is 448K for the known-affected flight controllers.

The firmware is actually flashed fine, this is just a messaging issue.

Currently affects SPRacingH7NANO, SPRacingH7EXTREME, SPRacingH7ZERO and likely any other EXST targets.

haslinghuis, 2020-08-12:
@hydra For F4 it is processing and parsing Device Feature, Internal Flash, Option Bytes and OTP Memory descriptors correct. So I cannot know if the descriptors returned are correct. The External Flash descriptor for STM32H750 which is hardcoded now should calculate to 4x128 = 512K according the code. So the value is wrong anyway, unless the descriptor has changed? How is the 448K presented in the descriptor? Without H7 hardware (and knowledge about EXST) I cannot really tell.

hydra, 2020-08-13:
the dfu descriptor is used to decide the start address and size of the region to be programmed.

haslinghuis, 2020-09-05:
@hydra can you answer my question please?

closed 2020-12-25 by stale[bot]