betaflight/betaflight-configurator #2153: Betaflight configurator shows no options to configure board.

tuskiomi, 2020-08-09:
If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
When betaflight configruator is open, there are no options to configure the slected board.
2. Include ways to reproduce the problem;
Open the program
Select any board from the flash program menu
voila, no settings.
3. Provide as much information as possible about your hardware and software, including:
There is no hardware involved in this process, except the hardware on which the program is running on, which is a standard pc running Windows 10 64-bit. An internet connection is present. Firmware has been downloaded.

haslinghuis, 2020-08-09:
Please explain what you mean by voila, no settings in detail

Did you see the message `Loaded release information for firmware from Github`?
Could you please attach the log from Configurator.

tuskiomi, 2020-08-09:
Here is the log:
```
2020-08-09 @ 20:13:27 -- Running - OS: Windows, Chrome: 80.0.3987.116, Configurator: 10.7.0

2020-08-09 @ 20:13:28 -- Loaded release information for configurator from GitHub.

2020-08-09 @ 20:13:34 -- Loaded builds information for jobs from build server.

2020-08-09 @ 20:13:36 -- Loaded release information for firmware from GitHub.

2020-08-09 @ 20:13:36 -- Loaded release information for firmware from GitHub.

2020-08-09 @ 20:14:00 -- Using cached builds information for jobs.

2020-08-09 @ 20:14:00 -- Using cached release information for firmware releases.

2020-08-09 @ 20:14:01 -- Using cached release information for firmware releases.

2020-08-09 @ 20:16:03 -- Using cached release information for firmware releases.

2020-08-09 @ 20:16:08 -- Using cached release information for firmware releases.

2020-08-09 @ 20:20:40 -- Using cached builds information for jobs.

2020-08-09 @ 20:20:41 -- Using cached release information for firmware releases.

2020-08-09 @ 20:20:41 -- Using cached release information for firmware releases.
```

And the explanation is rather simple. The only configuration option that is visible is the baud rate, the board and the firmware version.

haslinghuis, 2020-08-09:
This is my view:

[image]

tuskiomi, 2020-08-09:
Where's the PID tuning rates? 
Where's the configuration for number of rotors?
Where's  the configuration for battery voltage warnings? 

It seems we're seeing the same thing.

haslinghuis, 2020-08-10:
Verify your COM port and click on connect.
[image]

If you have problems connecting please try a verified USB data cable. And it that doesn't work look here

[image]

haslinghuis, 2020-08-10:
Please verify your COM port and click on Connect.
[image]

If you have problems connecting please try a verified USB data cable. And it that doesn't work look here

[image]

tuskiomi, 2020-08-11:
I don't have a board. I want to configure before I connect.

haslinghuis, 2020-08-11:
This is currently only possible using SITL:

https://github.com/betaflight/betaflight/tree/master/src/main/target/SITL

McGiverGim, 2020-08-11:
@tuskiomi this is not the way the Configurator works. You need to connect your FC to configure it. If you have more user questions, please use rcgroups or the official Betaflight Slack chat. For development issues you can continue using this github.

closed 2020-08-11 by McGiverGim

tuskiomi, 2020-08-11:
> @tuskiomi this is not the way the Configurator works. 

Why not?

McGiverGim, 2020-08-12:
There are several reasons, but basically, the Configurator needs to connect to the FC to know what version do you have, what features your FC support, some users compile it's own firmware with more/less features added, etc. so without connecting and asking to the FC we don't know what to show in the Configurator.

mikeller, 2020-10-26:
@tuskiomi: 

> Why not?

Mainly because nobody has implemented this yet. Betaflight is a community driven open source project, and all functionality is only there because somebody volunteered to spend their free time implementing it. If you think this is needed functionality, then feel free to volunteer your time to implement it.