betaflight/betaflight-configurator #2155: Idea for OSD:  RED FONT for use Night Flights?

diveflyfish, 2020-08-17:
Hello All.  I just wanted to share an idea that perhaps has already been contemplated.  With the new Low Lux Cameras such as the Foxeer Micro Cat II, Nano Toothless, Caddx Ratel et al, the ability to fly a night in grey scale or color is now reality.  However, the OSD font is default in White.  Is there any means to have the color shift to RED font when a low lux camera switches to low level Grey scale or upon Beta flight choice as per local time of day etc?  This is a minor issue to be sure.  It would be however really cool to maintain Night vision eyesight.  Thank you for your consideration of this.  
Very Best Regards,
James E. Hamman 
hfdresearch@icloud.com

mikeller, 2020-08-17:
@diveflyfish: Good idea. Unfortunately not possible with the currently used MAX7456 based on-board OSD hardware, as it supports overlaying black and white only.
**But** you can use `displayport_max7456_blk` and `displayport_max7456_wht` (also available in the 'OSD' page in CMS) to adjust the brightness of black and white, and reducing the 'white' brightness should give you better results when flying in low light conditions.

diveflyfish, 2020-08-17:
Thank you so very much!
Have a great week.
Very Best Regards,

James E Hamman DDS

closed 2020-10-04 by stale[bot]