betaflight/betaflight-configurator #2161: Betaflight wont open on Windows 10

el-kwadro-fpv, 2020-08-23:
Hello,

i configured and flashed a F722 flight controller. After that i want to start up Betaflight.exe again but it wont open up anymore. Task manager shows that there is a Task open, but nothing happens anymore.

I uninstalled and reinstalled BF 10.7.0 severeal times. It doesnt help.

It worked allways flawless but from one moment to another its not working anymore.

haslinghuis, 2020-08-23:
Try solution posted here: https://github.com/betaflight/betaflight-configurator/issues/2063

el-kwadro-fpv, 2020-08-24:
> Try solution posted here: #2063

Thanks for that hint, but unfortunately it wont work out.

mikeller, 2020-08-24:
@el-kwadro-fpv: What GPU / graphics card do you have in this PC, and what driver version for it is installed?

el-kwadro-fpv, 2020-08-24:
> @el-kwadro-fpv: What GPU / graphics card do you have in this PC, and what driver version for it is installed?

@mikeller Nvidia Geforce GT 650M Version 376.54

haslinghuis, 2020-08-24:
Maybe information below does apply to you? Are you using the drivers supplied by the OEM or did you update from Nvidia itself?

`"This is a reference driver that can be installed on supported NVIDIA notebook GPUs. However, please note that your notebook original equipment manufacturer (OEM) provides certified drivers for your specific notebook on their website. NVIDIA recommends that you check with your notebook OEM about recommended software updates for your notebook. OEMs may not provide technical support for issues that arise from the use of this driver."
`

el-kwadro-fpv, 2020-08-24:
> Maybe information below does apply to you? Are you using the drivers supplied by the OEM or did you update from Nvidia itself?
>
> `"This is a reference driver that can be installed on supported NVIDIA [...]

until a couple hours the OEM drivers where installed. I made an update just right now to try out.

BF used to work until i flashed the FC to BF version 4.2.2. Now it wont start up anymore

haslinghuis, 2020-08-24:
Please uninstall BFC 17.0 and manual remove `C:\Users\USERNAME\AppData\Local\betaflight-configurator` and install again to see if this helps.
Also is there a posibility to switch the GPU to use the Intel Graphics Adapter?

el-kwadro-fpv, 2020-08-24:
> Please uninstall BFC 17.0 and manual remove `C:\Users\USERNAME\AppData\Local\betaflight-configurator` and install again to see if this helps.

Allready did this about 10 times. Switched to 10.6, 10.7 RC1 and RC2 and back to 10.7. No BF works anymore.

> Also is there a posibility to switch the GPU to use the Intel Graphics Adapter?

Yes it is. I allready deactivated the GPU but it's still not working for me.

el-kwadro-fpv, 2020-08-24:
> Please uninstall BFC 17.0 and manual remove `C:\Users\USERNAME\AppData\Local\betaflight-configurator` and install again to see if this helps.

deleting the folder manually worked out! Thank you so much.
I feel pretty stupid, because i just deleted the remaining files in the installation folder.

Can't tell you how glad i am right now! THANKS.

closed 2020-08-24 by el-kwadro-fpv