betaflight/betaflight-configurator #2181: Gyro Filter Slider stays active when all lowpass turned off.

KAJE1977, 2020-09-22:
I have noticed when i turn off the gyro filter completely that the slider stays active and shows 1.05. All the buttons are clicked to the off position from the slider being originally at 2x. Also in the lua script it shows a min value of 0 and a max value of 1000. Im not sure how to check to see if the filter is dynamically jumping from 0 to 1000 or if it is truly off.

McGiverGim, 2020-09-29:
Look at the CLI values with a diff, they will always show if the filter is enabled or disabled (zero).

haslinghuis, 2020-09-29:
Tested and seems behavior has changed:

10.7 the Gyro filter slider is disabled when disabling all gyro (lowpass / notch / rpm) filter settings.
10.8 the Gyro filter slider is not disabled when doing the same nonetheless configurator is saying the sliders are disabled.

Some findings (10.8):

- On 10.7 only the Gyro filter slider would be disabled. On 10.8 the two sliders are on (visable) or off (visible or hidden)
- Fiddling with the sliders in the D-Term Lowpass Filters box will display / hide the sliders
- Enabling D-Term Lowpass 1 dynamic min cutoff freq + D-Term Lowpass 2 Cutoff Freq will turn both sliders back on

McGiverGim, 2020-09-29:
Is clear that the sliders do not work well.

My opinion always was that this sliders needed to be implemented in the firmware side, and the Configurator only show them. In this way all works in an easy way. But the users wanted them and we add them in a bad way, letting the Configurator try to "guess" the correct slider position based on the final data, thing that is not always possible.

Plus some fix added between versions seem to have broken it more. :(

KAJE1977, 2020-09-30:
Yes the filter was definitely off it just showed the range and kept the slider on for whatever reason. It also showed a range from 0 to 1000 in the lua scripts. I just went in to cli and set all numbers to 0. It now shows all 0 for range. The slider is still shown as active but it now says 0 instead of 1.05.

KAJE1977, 2020-09-30:
Sorry not active. Bad wording. The slider is still there but shows 0 and not 1.05.

haslinghuis, 2020-09-30:
@KAJE1977 How I see the use of the sliders:

The sliders are for the majority of less experienced users to enable them to tune the quad with some guidance. Keeping them in sync after changing individual settings was never the intention afaik - agreed still needs improvement.

As long there is no solution please use the sliders only for initial setup or as long you don't touch individual settings.
If you want to tune like an expert ignore the sliders and verify in the cli.
Also OSD or LUA script could change various settings which the configurator would not know about.

Weeperr, 2020-10-02:
I've noticed an issue with the filter sliders also in BF 4.2.3. If I change them from defaults, hit save and then switch PID profiles within the configurator and then go back to the previous profile the slider position will change and then if go back again to the other profile then those sliders will change also. It appears that the only slider position that will save is a 1.0 for both sliders

TheIsotopes, 2020-10-07:
I can confirm ... only Gyro Lowpass 2 Cutoff is deactivated and it looks like this:

[image]

PR #2233 (2020-10-19): Fix tuning sliders
Fixes #2181
closed 2020-10-19 by mikeller