betaflight/betaflight-configurator #2184: Changing motor_output_limit from % to max voltage.

zeusalex, 2020-09-29:
I was analyzing how the variable “motor_output_limit” works and I thought that we could improve its functioning. 
For example, if I have motors for 4S but I'm running on 6S, I would use 66% ... but this is only valid when the battery is at 100%.

These motors were designed to work at 16.8v but the system is working at 25.2v,.. I need to limit it to 66% only when the battery is 100% charged.

Right after a brief use, the battery can now be at 4v/c => 24v ... In this case, my new limit is16.8 / 24 => 70% ... And it keeps increasing. .. with 3.8v / c we are at 16.8 / 22.8 = 74% ...

I believe that someone must have already thought about it, but instead of making a motor_output_limit %, why don't we use a maximum voltage?

With this we would always guarantee that the electric motors will always be working at 100% of the designed voltage.

McGiverGim, 2020-09-29:
I don't know if the voltage will be fiable/fast enough for that... others with more knowledge can confirm if viable. But is better that you post your feature request to be discussed in the firmware part, that is the place where this change needs to be done. This is the link: https://github.com/betaflight/betaflight/issues

I can move it to there, but we have a template that you need to fill so better to move there and copy your text in te correct places of the template.

haslinghuis, 2020-09-29:
You can already do this with `auto_profile_cell count` and `motor_output_limit` using pid profiles, just make sure you set maximum cell voltage to 4.4 instead of 4.2 to detect the cells right. Percentage is calculated by taking the 6S motor KV (what you want) and divide it by the actual KV of your 4S motor (what you have) and multiply by 100 to get the motor  percentage. Example of KV used on 4S motors in my table:

6S KV | 4S KV | Percentage
-- | -- | --
1750 | 2750 | 64
1750 | 2800 | 63
1750 | 3200 | 55
1750 | 3600 | 49
1750 | 3750 | 47
1750 | 3800 | 46

Before setting this up make sure you're other components are rated for 6S also.

closed 2020-09-30 by zeusalex