betaflight/betaflight-configurator #2191: RFC: Code quality tools

chmelevskij, 2020-10-02:
## Problem

After looking at some PRs I've noticed that there is some inconsistencies in the code and fair amount of comments about spacing, style and formatting. While there is Sonarcloud which seems to catch such thing it would be nice to have code in the right shape before it even hits PR.

## Propsed solution

Introduce pre-commit hooks which will lint and format the code accordingly. Taking gradual approach and adding new tools as code evolves into modules/vue.

Proposed plan:

- [ ] 1. Introduce eslint, with warnings and globals. https://github.com/betaflight/betaflight-configurator/pull/2189
- [ ] 2. fix all the warnings, make those error level.
- [ ] 3. add stricter formatter, E.G. `prettier`
- [ ] 4. adjust rules with the team

mikeller, 2020-10-04:
While automatic linting / formatting may be a great tool in a professional environment where people are contractually forced to own changes that the system makes on their behalf, I am not a fan of automatic linting for community driven projects - in my experience this will just result in a lot of code that is poorly tested in the way it is submitted, and authors that will go 'nah, I don't know what this does and how it is supposed to work, this is not what I wrote'.

chmelevskij, 2020-10-04:
But isn't what sonar cloud doing this as 3rd party service anyways? The only difference being that you get feedback when submit PR, and from my short experience on this project, you get pushback form the reviewers like: _put space here_.

chmelevskij, 2020-10-04:
> this is not what I wrote

I can see your point. `prettier` especially has quite a opinionated formatting. If there is wish to allow for some wiggle room around formatting it's possible to use just `eslint` only, quite often it has looser formatting rules.

mikeller, 2020-10-06:
If we allow the wriggle room I do not really see why we should move away from using SonarCloud for this purpose, since this already integrates with GitHub in a nice form, and is thus available to everybody - getting eslint to integrate with GitHub and make the results easily accessible seems to be a lot harder.

chmelevskij, 2020-10-06:
`eslint` is installed in the same way as `gulp` or any other package in `package.json` so you can run it on your machine, right now with `SonarCloud` it seems that only core team can get the benefits of it on own machines. `eslint` seems more accessible if you ask me.

McGiverGim, 2020-10-06:
That's not totally true, we can "export" the SonarCloud rules and let them accesible for everybody. But the rules are dynamic and changing with the new "programming" tendencies, so I don't know if this is a good idea.
For this I use a connection with SonarCloud directly, but as you say, I'm a member and no everyone can do this.

mikeller, 2020-10-06:
@chmelevskij: Are you going to run `eslint` on all pull requests and post the results to GitHub then? Because I am certainly not willing to do this for every pull request before I review it, just in case somebody has not gone through the trouble of installing a linter and running it before opening a pull request.

chmelevskij, 2020-10-06:
linter is installed the same way as everything else in `node_modules`. So if you are able to do `yarn start` you will be able to run `yarn lint`.

First step is to integrate `eslint` with pre-commit hook which works on your machine (not Github GUI since it doesn't run any git hooks). I've done that in https://github.com/betaflight/betaflight-configurator/pull/2189 it also provides `yarn lint:fix`.

General workflow will change for contributors from:
```
make changes -> commit -> push -> make PR -> Sonar Cloud -> fix the text locally -> commit -> push -> sonar cloud...
```

To something like
```
make changes -> trigger commit -> get lint error -> lint:fix or manual fix -> commit -> push -> pr 
```

mikeller, 2020-10-11:
@chmelevskij:

```
make changes -> trigger commit -> get lint error -> lint:fix or manual fix -> commit -> push -> pr 
```

Which part of this will make it possible for me as a reviewer to verify that a linter has been run for any given pull request? As I said, there is no interest from any reviewers in this project to do more manual work for every pull request.

> Now there is reliance on contributors or maintainers to either pull and fix or add suggestions, this removes that.

This is simply wrong - SonarQube will prevent mergeing of any pull requests that fail to meet the defined code quality goals - this includes coding style _and_ problem analysis.

chmelevskij, 2020-10-11:
> Which part of this will make it possible for me as a reviewer to verify that a linter has been run for any given pull request? 

Pre commit hook and adding it to GH actions or travis.

McGiverGim, 2020-10-11:
One thing about SonarQube: it includes code style when it can help to prevent bugs, but I have not found rules about for example add or not an space after parenthesis, comma, etc.

mikeller, 2020-10-11:
@McGiverGim: Ok, that makes sense to me. The integration of eslint into GitHub seems to be a lot worse than what we get for free from SonarCloud, but we can always consider to use eslint to enforce rules that we cannot enforce in SonarCloud.

haslinghuis, 2020-10-12:
Ultimately developers want to have sonar / eslint issues raised and resolved local before making the commit. An instruction to use an `eslint` plugin in your editor / IDE could be sufficient.

dellch, 2021-12-05:
Is the next step for this issue to fix all warnings?

chmelevskij, 2021-12-19:
Closing as we added the eslint. Will stay away from the prettier at least for now. Adding more rules to match sonar cloud as we go.

closed 2021-12-19 by chmelevskij