betaflight/betaflight-configurator #2196: Arming the quad when it is not suppose too

Ric-Alves, 2020-10-04:
Hi, I was testing new setup on modes tab and testing each one. I tough if I don't turn on the "safety" switch on motor tab, the motors will not spin at all, even I had the props on (a big mistake I know).
When I was testing the arm switch and the pre-arm switch, the motors start to spin...
I notice if I only activate arm switch the arm mode goes red with the message "(disable)", as suppose.
But if I activate the Pre-arm switch and next the arm switch, like I would do to fly, the arm mode goes yellow and the motors start to spin!
I thought that with the usb cable connected, betaflight configurator connected, the motors would only spin with the "safety" switch activated, at motors tab.

Is this a normal behavior? It could be danger to someone that doesn't know this.

It seams to me a configurator issue, or it would be a Betaflight issue?

Link youtube https://youtu.be/ABIy5fBAxbQ
Betafligh configurator 10.7.1
Betaflight : 4.2.3

`# 

# diff all

# version
# Betaflight / STM32F405 (S405) 4.2.3 Sep 20 2020 / 20:42:51 (2696b7c88) MSP API: 1.43
# config: manufacturer_id: DIAT, board_name: FURYF4OSD, version: 3a35e73b, date: 2019-09-30T05:46:12Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name FURYF4OSD
manufacturer_id DIAT
mcu_id 003d002f374d500320313048
signature 

# name: Alves

# feature
feature -AIRMODE

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200

# aux
aux 0 0 3 900 1125 0 0
aux 1 13 1 1925 2050 1 0
aux 2 26 0 900 1125 0 0
aux 3 28 2 1375 1600 0 0
aux 4 31 3 1800 2100 1 0
aux 5 31 1 1275 1400 1 0
aux 6 35 2 950 1150 0 0
aux 7 36 1 1700 1825 0 0
aux 8 39 0 1850 2100 1 0
aux 9 39 1 1450 1550 1 0
aux 10 39 3 1800 2100 1 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 25 100 200 400
vtxtable powerlabels 25 100 200 400

# master
set gyro_lowpass2_hz = 375
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 90
set dyn_notch_max_hz = 360
set dyn_lpf_gyro_min_hz = 300
set dyn_lpf_gyro_max_hz = 750
set acc_hardware = NONE
set acc_calibration = -39,9,99,1
set min_check = 1010
set max_check = 1990
set rssi_channel = 16
set rssi_src_frame_errors = OFF
set rc_smoothing_auto_smoothness = 7
set blackbox_p_ratio = 64
set dshot_idle_value = 320
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set bat_capacity = 680
set vbat_warning_cell_voltage = 355
set beeper_dshot_beacon_tone = 5
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 1
set yaw_deadband = 1
set pid_process_denom = 2
set osd_warn_batt_not_full = OFF
set osd_warn_visual_beeper = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_warn_rssi = ON
set osd_cap_alarm = 700
set osd_logo_on_arming = FIRST_ARMING
set osd_tim2 = 1281
set osd_vbat_pos = 6594
set osd_rssi_pos = 6201
set osd_tim_2_pos = 6614
set osd_throttle_pos = 6177
set osd_vtx_channel_pos = 6645
set osd_current_pos = 6625
set osd_mah_drawn_pos = 4554
set osd_craft_name_pos = 6209
set osd_battery_usage_pos = 2536
set osd_disarmed_pos = 6187
set osd_esc_rpm_pos = 4247
set osd_stat_max_spd = OFF
set osd_stat_battery = ON
set osd_stat_bb_no = OFF
set osd_stat_max_esc_rpm = ON
set osd_stat_max_fft = ON
set debug_mode = GYRO_SCALED
set vtx_band = 3
set vtx_channel = 3
set vtx_power = 1
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5665
set gyro_1_align_yaw = 1800
set gyro_rpm_notch_q = 700
set name = Alves

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 105
set dyn_lpf_dterm_max_hz = 255
set dyn_lpf_dterm_curve_expo = 7
set dterm_lowpass2_hz = 225
set vbat_pid_gain = ON
set anti_gravity_gain = 4000
set iterm_relax_type = GYRO
set yaw_lowpass_hz = 100
set throttle_boost = 7
set throttle_boost_cutoff = 25
set p_pitch = 48
set i_pitch = 72
set d_pitch = 43
set f_pitch = 84
set p_roll = 44
set i_roll = 68
set d_roll = 40
set f_roll = 79
set p_yaw = 47
set i_yaw = 72
set f_yaw = 79
set d_min_roll = 26
set d_min_pitch = 29
set ff_spike_limit = 70
set ff_smooth_factor = 20
set idle_min_rpm = 20

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 71
set tpa_breakpoint = 1550

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# `

McGiverGim, 2020-10-04:
Are you able to reproduce it each time (without props of course)?
It seems to me more a firmware issue, but it's not 100% sure.

Ric-Alves, 2020-10-05:
> Are you able to reproduce it each time (without props of course)?

As you can see in the video, each time I tried to reproduce it, even to do the video, it occurs.

> It seems to me more a firmware issue, but it's not 100% sure.

I had that doubt too. If someone confirm that this is a firmware issue, I post as issue on firmware github.

McGiverGim, 2020-10-05:
I can't do a test right now, but is easy to see.

The Configurator sends an MSP command to enable arm when the switch of the motors tab is enabled.

So to check this problem:
- If this only happens when there is a pre-arm switch enabled, the problem is in the firmware. The Configurator knows nothing about this mode, so for the Configurator is the same to have it or not.
- If this happens too without pre-arm switch, then maybe the Configurator sends the command to enable arm by error.

Can you test if this happens too removing the pre-arm mode for a test?

Thanks! This is a safety issue, so is important to fix it.

Ric-Alves, 2020-10-05:
Ok this is strange.

Yesterday I reproduce the issue as many times as I wanted, it even let me
do the video two times.

Today I can't reproduce it....

I don't know if was related but it happens after update the bf from 4.2.0
to 4.2.3.

Should I report as issue on betaflight firmware github?

McGiverGim, 2020-10-05:
I don't know what to say. Is a bug that needs to be fixed, but without a way to reproduce it will be difficult. Maybe it is related to another action that you did in the craft or some save at some tab in the Configurator.