betaflight/betaflight-configurator #2214: OSD problems

KAJE1977, 2020-10-10:
I had a recent issue with all 5 of my flight controllers. I am using the nightly builds of the configurator. I was using the latest nightly and had uploaded images to the osd no problem. It was only when i uploaded a different font i ran into a problem. I had thick black bars across my whole image in fov feed and small random pixels of my video were showing through. Hard to explain. But i kept flashing previous nightly builds until i got to the version before 2020104.967. I think there was alot of osd refactoring in this version. Anyway i could upload an image completely fine but the font is what screwed it up on all of my flight controllers. Another wierd issue too was i tried flashing over to emuflight and the problem carried over until i went back to betaflight and went to the configurator version before the one i posted here and it fixed it. I dont understand at all why. Just an observation. If this is user error them i apologize in advance. Thank you.

McGiverGim, 2020-10-11:
If I understood well, the OSD goes mad when you upload a new font. I will try to reproduce, but until Tuesday I'm out.
Someone more can confirm the issue?

Asizon, 2020-10-11:
Tested @McGiverGim , definitelly it is a issue in osd code, using latest 10.8.0 build and old 4.3 build,it looks like that:

[image]

KAJE1977, 2020-10-12:
Yes exactly on everyone of my flight controllers and the issue carried over to a different firmware " emuflight ". The only fix was to go back to the release before the one i posted i believe.

KAJE1977, 2020-10-12:
Fixed right away. Thank you so much

closed 2020-10-12 by KAJE1977