betaflight/betaflight-configurator #2219: betaflight configurator 10.7 freezes

ryansmall75, 2020-10-12:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide as much information as possible about your hardware and software, including:
- what hardware / software you are using;
- the version of all the software used;
- how things are connected / wired up.
4. Remove this Text :).

Asizon, 2020-10-12:
@ryansmall75 please try that:
https://github.com/betaflight/betaflight-configurator#graphics-issues

ryansmall75, 2020-10-14:
Not a graphics issue, the program freezes as soon as i try to click any of the tabs...
FXAA was already off

Asizon, 2020-10-14:
Yes, if you can see a box in setup tab, it is a graphic issue.

ryansmall75, 2020-10-14:
Ah yes, you are correct. I thought BF worked with my other flight controllers yet. I cannot use 10.7 at all.

Asizon, 2020-10-18:
This issue can be closed

closed 2020-10-18 by McGiverGim