betaflight/betaflight-configurator #2240: Problem to re-run Azure for Windows/Android when Android failed

McGiverGim, 2020-10-20:
At some build, like this https://dev.azure.com/Betaflight/Betaflight%20Nightlies/_build/results?buildId=1087&view=results the Windows/Android build has failed.
The Windows build do it for Windows and for Android, one after the other. The Windows build has finished but the Android one has suffered a temporary problem with some dependencies to download.
If we re-run the execution, it fails always because the Windows artifact was generated the first time and can't be overwrited.
What is the best solution here:
1. Divide the execution into two: I think is the best and elegant solution if Azure let us (I don't know the requisites/limits of this free tool).
2. Delete the old artifact if exists before uploading the new.

@mikeller what do you think?
@Asizon you did the Android part. Thoughts?

Asizon, 2020-10-20:
@McGiverGim yes good point, anyway azure always abort release step if one of the plataforms fails, then I think we cant do anithing making a separate build for Android, if it fails releases never executed.

McGiverGim, 2020-10-20:
The difference is that if we apply one of the solutions, we can re-run them, and if the problem was temporary, the build will finish. Now we can't.

I have seen that for solution 2, we can simply "ignore" the error produced if it exists, in the `PublishPipelineArtifact` adding a continueOnError: true` control option:
https://docs.microsoft.com/en-us/azure/devops/pipelines/process/tasks?view=azure-devops&tabs=yaml#controloptions

But again, I prefer to generate a new VM only for the Android build.

McGiverGim, 2020-10-20:
According to this page: https://github.com/marketplace/azure-pipelines

> Azure Pipelines offers cloud-hosted pipelines for Linux, macOS, and Windows with 10 free parallel jobs and unlimited minutes for open source projects

If we can make use of 10 parallel jobs, without limit in minutes, is clear that the best solution is let Android to use it's own VM.

mikeller, 2020-10-20:
I'm all for 1.

closed 2020-10-21 by mikeller