betaflight/betaflight-configurator #2243: run-script-os ignores parameters

McGiverGim, 2020-10-20:
Until now we can build a, for example, win64 version with `yarn release --win64`.

Since we add the `run-script-os` the `--win64` parameter is ignored and the `--win32` version is generated by default.

It can be build with a `yarn gulp release --win64`, because it does not use the `run-script-os` but I think is good to maintain the parameters for all of the yarn scripts. 

@atomgomba some idea about how to fix it?

atomgomba, 2020-10-21:
Fortunately I have a Windows machine to look into this now! It's unfortunate `run-script-os` has this limitation and a CI pipeline didn't fail when this change was merged. It turned out that there is an [existing PR](https://github.com/charlesguse/run-script-os/pull/32) for this issue opened last month. It would be best if the author reviewed and merged it so I will try to make contact to see if it's gonna work.

McGiverGim, 2020-10-21:
It seems the developer has forgotten this two PR ;)

Ok, let's wait to see if he can merge and publish a new version.

atomgomba, 2020-10-22:
Looks like with the merged changes the arguments are now passed correctly:

`Building for platform(s): win64.`

McGiverGim, 2020-10-22:
Perfect! Thanks for the fix  ;)

closed 2020-10-25 by McGiverGim