betaflight/betaflight-configurator #2246: Android version is broken in master

McGiverGim, 2020-10-21:
It seems that it does not load since some changes of libraries.

@WalcoFPV does cordova support loading libraries like we are doing? it supports libraries made for node?

I have made some changes in latest versions, and some more are pending. We need to find a way that works for both.

Is a pity that now that we have get rid of Chrome, we need to take into account Cordova, that is more difficult to detect.

What is the best way to debug this things? In Node we have the developers tools, but in Android I'm not too sure how to do it...

McGiverGim, 2020-10-21:
It seems it fails since https://github.com/betaflight/betaflight-configurator/pull/2230 but maybe others pr later make I'm not too sure if others pr later have the same problem.

McGiverGim, 2020-10-21:
Problem found, Android loads it in other main:
https://github.com/betaflight/betaflight-configurator/blob/7d3306aecbe09243605cb55f2b4d195fba2dfbdf/src/main_cordova.html#L43

I will fix it tomorrow...

mikeller, 2020-10-21:
@McGiverGim: Now that we dropped support for the Chrome Web version, we might be able to look into unifying the two main html files - this will avoid this pitfall for the future.

McGiverGim, 2020-10-21:
I don't know where to start to do it... The Cordova thing is too unknown for me at this moment 😁

WalcoFPV, 2020-10-22:
> @McGiverGim: Now that we dropped support for the Chrome Web version, we might be able to look into unifying the two main html files - this will avoid this pitfall for the future.

Technically, Cordova doesn't required a special main html file. But because some devices don't have compatible webview, we had to have this `main_cordova.html` file, coded to work with old webview. It is used only to inform the user that the configurator can't be used without updating the webview app. If everything is good with the device, `main_cordova.html` will redirect to the other main html file. So if we unify the two main html files, the function of the cordova main file doesn't exist no more

WalcoFPV, 2020-10-22:
> @WalcoFPV does cordova support loading libraries like we are doing? it supports libraries made for node?

Yes cordova supports like nwjs, libraries made for node

> What is the best way to debug this things? In Node we have the developers tools, but in Android I'm not too sure how to do it...

You can use almost the same developers tools with Android. You have to enable usb debugging on your phone. Then, with a debug version of the configurator installed on your phone, plug your phone to your computer. And on your computer, go on Chrome and go to chrome://inspect. Normally, you will find your Android device

McGiverGim, 2020-10-22:
Thanks for the explanations!
I think at some moment we need to do a `developing_android.md` document, explaining how to open it with Android Studio and things like this. At this moment I do all with the `yarn` command and all is slowwwwww :)

WalcoFPV, 2020-10-22:
I don't know if it's really useful to use Android Studio. I only used it to install the Android SDK. And also, you can accelerate the build of the app. You just have to build the app one time. After that, add the option --skipdep in the yarn command to ignore the rebuild of the cordova depedencies.

McGiverGim, 2020-10-22:
ahhhh perfect, I didn't know of this option. As I say we need to add all of this info to a document.

I have tested the patch fixing the script, now it loads but for some reason it does not find the messages. Does Cordova customize the translation system? I will try to debug it but maybe you know the answer in advance...

WalcoFPV, 2020-10-22:
When building Android app, gulp renames the `_locales` folder into `i18n`. I don't know why but when the folder is named  `_locales`, the app doesn't work. If i18n library doesn't find the folder containing the locales files, this could be the issue

McGiverGim, 2020-10-22:
The `locales` is renamed into `_locales` because we needed to maintain compatibility against Chrome app, that needed this folder. Now that we have removed the Chrome app support we can let the folder with the original name.
But in theory, for this load we use the i18n/i18next library that is the same for windows/android and under windows is working.

McGiverGim, 2020-10-22:
@WalcoFPV thanks to your instructions,debug accomplished, the error is as I suspected, the `require` is not defined:

>Uncaught ReferenceError: require is not defined
    at localization.js:23
    at getStoredUserLocale (localization.js:208)
    at Object.i18n.init (localization.js:21)
    at Channel.onDeviceReady (main_cordova.js:232)
    at Channel.fire (cordova.js:868)
    at cordova.js:225

Require is the way for node to import modules. It can use import too, but the require has less requisites (for example it can be used out of a module).

What is the solution for this?
- Use import at all places, this will need a refactor of some things
- Cordova can work with `require` with some change...

mikeller, 2020-10-22:
@WalcoFPV:

> Technically, Cordova doesn't required a special main html file. But because some devices don't have compatible webview, we had to have this main_cordova.html file, coded to work with old webview.

Ah yes, now I remember. I think the confusion comes from a less-than-ideal file name here - renaming this to `main_cordova_unsupported_webview.html` or similiar would make this more obvious.

McGiverGim, 2020-10-23:
Only to let here the state of the Android build, I have tried to change to `import` the inclusion of the modules in the `localization.js` file, without luck. 
Someone has suggested maybe to use `rollup` over them, but at this moment I don't know what to do with this.
If someone with more knowledge of modules want to attack this I will be very happy to learn 😄

closed 2020-10-30 by mikeller