betaflight/betaflight-configurator #2253: Error message when starting configurator in debug mode

mikeller, 2020-10-23:
Reproduction:
- start configurator with `yarn start`

Expected behaviour:
- the configurator window opens

Actual behaviour:
- an error message is displayed for a few seconds:
[image]
- the error message closes and the configurator window opens

System:
Ubuntu 20.04

haslinghuis, 2020-10-23:
I needed a new `yarn install` last week, not aware about changes after that

McGiverGim, 2020-10-23:
These are the Vue tools. Maybe not available under Linux? I need to check, my Linux VM is not booting, I will try when I can fix the problem.

haslinghuis, 2020-10-23:
I use `yarn gulp debug` (`yarn start` looks identical)

McGiverGim, 2020-10-23:
@haslinghuis this Vue tools were merged 5 days ago. Do you have tested after that?

haslinghuis, 2020-10-23:
Not sure, going to test when I'm back home.

haslinghuis, 2020-10-23:
Actually `yarn gulp debug` still works.
Using `yarn start` I get 
```
yarn run v1.22.5
$ run-script-os
/bin/sh: run-script-os: command not found
error Command failed with exit code 127.
info Vistit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
```

So to let `yarn start` work again I needed `yarn install` command again.
It has some errors but starts the configurator.
```
yarn run v1.22.5
npm WARN lifecycle The node binary used for scripts is /tmp/yarn--16034758765567-0.9612878228777699/node but npm is using /usr/bin/node itself. Use the '--scripts-prepend-node=path' option to include the path for the node binary npm was executed with.
```

mikeller, 2020-10-24:
Na, even after a `yarn` I still get the error message.

PR #2256 (2020-10-25): Updated dependencies to fix error message when starting in debug mode.
Fixes #2253.
closed 2020-10-25 by mikeller