betaflight/betaflight-configurator #2257: Configurator not starting in debug mode

mikeller, 2020-10-24:
Reproduction:
- start configurator in debug mode with `yarn start`;

Expected behaviour:
- a configurator window opens

Actual behaviour:
- the following error message appears:
[image]

McGiverGim, 2020-10-24:
It says localisation used. Where we have chrome localisation? I tested it in my system before my pr. Let me do tomorrow more tests.

McGiverGim, 2020-10-24:
Maybe you have the _locales folder remaining? The debug mode did not erase old files before building.

haslinghuis, 2020-10-24:
After deleting the remaining _locales folders it works!

mikeller, 2020-10-25:
@McGiverGim: Yes, you are correct, deleting `./debug/betaflight-configurator/linux64/_locales` fixed this.

closed 2020-10-25 by mikeller