betaflight/betaflight-configurator #2259: Save configuration as image set

wyfinger, 2020-10-25:
Sometimes I want to have a FC setup on hand in a simple and understandable form, but betaflight-configurator does not allow you to open json for viewing without the FC. Usually I take a screenshot of the configurator tabs and save them as a set of pictures. But my laptop has a small screen and the configurator settings pages do not fit in height, I have to scroll.

Is it possible to add a button to automatically save a set of screenshots (in full height)?

I've been experimenting with `html2canvas.js` and in my opinion it's possible, below is an example of code:
```javascript
document.addEventListener("DOMContentLoaded", () => {
	$('a.savepng').click(function () {
		html2canvas(document.body).then(function(canvas) {           
		//debugger;
		var a = document.createElement("a");
			document.body.appendChild(a);
			a.href = canvas.toDataURL('image/png');
			a.download = "screenshot.png";
			a.click();
			document.body.removeChild(a);
		});
	});
});
```
My experience is not enough to implement to the end (flipping through tabs). Perhaps more experienced developers will find the idea interesting.

mikeller, 2020-10-25:
@wyfinger:

Taking screenshots of the configurator UI is not a good way to persist the state of the Betaflight firmware - not all settings are represented in the configurator UI.

Using a CLI `diff` is a much better way, please read up on it:

https://www.propwashed.com/betaflight-backup-restore/
https://oscarliang.com/betaflight-cli-explained/

wyfinger, 2020-10-25:
@mikeller

Of course, but I often don't remember the exact name of the parameters, but I remember where it was in the configurator. Those. this is for clarity.

closed 2021-01-08 by github-actions[bot]