betaflight/betaflight-configurator #2269: BETAFLIGHT EXECUTABLE IS INFECTED

ylectron, 2020-11-06:
Adware.Dotdo.Z.Lokq | Filseclab
-- | --

McGiverGim, 2020-11-06:
It is not. We have several false positives with each version. Upload it to some web like https://www.virustotal.com/gui/ and you will se that the vast majority of antivirus approve it.

You can build it by yourself too if you are unsure.

McGiverGim, 2020-11-06:
Here you have the list of 60 antivirus that have analyzed the executable: https://www.virustotal.com/gui/url/eb04e73e0dbb3441755f8df6e13a2f65bbaa297e8da4f3110de280acfc8c828a/detection

mikeller, 2020-11-06:
Big part of this is that a lot of virus scanners (including Microsoft's) require an ~extortion fee~ registration of applications with them in order to not be tagged with all kinds of problems...

skywinder, 2020-11-15:
I think you can  close it as invalid, since there even no any valid references in the description.

@ylectron please specify the source where did you download  executable file

ylectron, 2020-11-16:
Definitely not closing it. Please download it by yourself from where you consider original source on github and make a test with the metascanonline engines ( https://metadefender.opswat.com/?lang=en). The previous version doesn't have any ADWARE, just the latest build for windows does.

McGiverGim, 2020-11-16:
I did, and only 1/37 detected "something" that I'm sure is a false positive:
[image]

You can believe 1 or 36 engines, as you want. As I say, you can build it yourself from the source code if are unsure.

mikeller, 2020-11-19:
False positive.

mikeller, 2020-11-19:
@ylectron: Writing in ALL CAPS is considered rude and bad style, please refrain from it in the future.

closed 2020-11-19 by mikeller