betaflight/betaflight-configurator #2273: Windows 10 Installer breaks Configurator Upgrade

derdanu, 2020-11-12:
Due to the change of _locales / locales the configurator can not be opened after installation if an old installation was present on the machine. 

Error Message: Failed to load extension from: C:\Programm Files(x86)\Betaflight\Betaflight-Configurator.Localisation used, but default_locale wasn't specified in the manitest. 

Manual Fix: Delete the _locales folder. 

Reproduce: 
Install betaflight-configurator-installer_10.7.0_win32
Upgrade with NIghtly 1,8,0 installer and start the configurator

McGiverGim, 2020-11-12:
In theory we execute the uninstall of the old version first. It seems it is not working as expected at least for the version 10.7. Did you see a message about this uninstallation?

derdanu, 2020-11-12:
Uninstallation was successful but only files in the root folder of the installation were deleted, no subfolder + files.

McGiverGim, 2020-11-12:
Then something went wrong with the uninstaller of version 10.7.

We have changed the installer system to fix some issues like this, we need to see if this happens always or only with some installations. We can try to delete the `_locales` folder from the installer as a workaround before installing.

haslinghuis, 2020-11-18:
Confirmed. Using latest nightly build on Windows have to remove `_locales` folder manually.

haslinghuis, 2020-11-18:
ISS script could be improved with
```
Function removeFolders()
  custActData = Session.Property("CustomActionData")
  foldersArray = Split(custActData, "|", -1)
  Set objFS = CreateObject("Scripting.FileSystemObject")

  For Each folder In foldersArray
    smallPath = RemoveTrailingBackslash(folder)
    If (objFS.FolderExists(smallPath)) Then
      objFS.DeleteFolder smallPath, true
    End if
  Next
End Function

Function RemoveTrailingBackslash(folderPath)
  If Right(folderPath, 1) = "\" Then
    RemoveTrailingBackslash = Left(folderPath, Len(folderPath) - 1)
  Else
    RemoveTrailingBackslash = folderPath
  End If
End Function
```
from: https://www.advancedinstaller.com/user-guide/qa-remove-file-or-folder.html

McGiverGim, 2020-11-18:
Maybe we can make use of this directive: https://jrsoftware.org/ishelp/index.php?topic=installdeletesection to remove only the `_locales` folder.

If I'm not wrong, you are removing all the files in the folder. This is dangerous, is not the first time that some user has installed Betaflight directly into de "program files" folder, without any sub folder.

Is clear that the old NSIS uninstaller is letting garbage, maybe not only the `_locales` folder. I don't know if it will be better to ask to uninstall and remove the old install folder first manually and then start clean with the new version.

haslinghuis, 2020-11-18:
If _locales is the only problem I guess it's safer to ask the user to manually remove it as for now it only impacts users using the nightly builds. Or use something like:

```
[UninstallDelete]
Type: files; Name: "{app}\_locales\*\messages.json
Type: dirifempty; Name: "{app}\_locales"
```
If the wildcard doesn't work:
```
[UninstallDelete]
Type: files; Name: "{app}\_locales\ca\messages.json
Type: files; Name: "{app}\_locales\de\messages.json
Type: files; Name: "{app}\_locales\en\messages.json
Type: files; Name: "{app}\_locales\es\messages.json
Type: files; Name: "{app}\_locales\eu\messages.json
Type: files; Name: "{app}\_locales\fr\messages.json
Type: files; Name: "{app}\_locales\gl\messages.json
Type: files; Name: "{app}\_locales\hr\messages.json
Type: files; Name: "{app}\_locales\hu\messages.json
Type: files; Name: "{app}\_locales\id\messages.json
Type: files; Name: "{app}\_locales\it\messages.json
Type: files; Name: "{app}\_locales\ja\messages.json
Type: files; Name: "{app}\_locales\ko\messages.json
Type: files; Name: "{app}\_locales\lv\messages.json
Type: files; Name: "{app}\_locales\nl\messages.json
Type: files; Name: "{app}\_locales\pl\messages.json
Type: files; Name: "{app}\_locales\pt\messages.json
Type: files; Name: "{app}\_locales\pt_BR\messages.json
Type: files; Name: "{app}\_locales\ru\messages.json
Type: files; Name: "{app}\_locales\sv\messages.json
Type: files; Name: "{app}\_locales\zh_CN\messages.json
Type: files; Name: "{app}\_locales\zh_TW\messages.json
Type: dirifempty; Name: "{app}\_locales"
```

haslinghuis, 2021-04-01:
See also: https://github.com/betaflight/betaflight-configurator/pull/2251

closed 2021-08-01 by haslinghuis