betaflight/betaflight-configurator #2274: macOS Big Sur crashes

AndreaCCIE, 2020-11-12:
## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
After having upgraded to macOS Big Sur, the configurator is very unstable. 
Is does crash often while on the PIDs tab, but it happens even on other tabs

2. Include ways to reproduce the problem;
Open the configurator, go to the PIDs tab and play with sliders/drop down

3. Provide as much information as possible about your hardware and software, including:
- what hardware / software you are using;
MacBook Pro 13 2018 (irrelevant)

- the version of all the software used;
macOS Big Sur 20B29
Betaflight Configurator both stable 10.7.0 and 10.8.0 v202011041226

Martivip, 2020-11-14:
i have the same issue with random crashes

mikeller, 2020-11-15:
Hmm, I can't reproduce the crashes with MacOS 11.0.1 and Betaflight configurator 10.7.0 on a VM, which is all that I have at the moment. Somebody with actual Apple hardware will have to see if they can reproduce / fix.
It _could_ also be a case of simply having to wait for the next release of NW.js, which should include fixes for any instabilities in MacOS Big Sur.

ljholland, 2020-11-16:
I'm having the same problem. It seems you can eventually get in and make changes but BF crashes or locks up.

Alex-g-by, 2020-11-17:
Same problem. Betaflight configurator 10.7.0 crashes after macOS update.
Mac mini (Late 2014), Big Sur 11.0.1

sn0opy, 2020-11-17:
Found this line an nwjs Helper diagnostic report:

> Wakeups:          45001 wakeups over the last 92 seconds (491 wakeups per second average), exceeding limit of 150 wakeups per second over 300 seconds

The moment this appeared, BFC just locked up and didn't let me do anything anymore. I doubt a full stacktrace would help here, but it looks like it's caused by nw.js not being fully compatible.

sn0opy, 2020-11-17:
There's a possible workaround as I was not able to reproduce the issue anymore after running the following command

```
codesign --remove-signature /Applications/Betaflight\ Configurator.app/Contents/Frameworks/nwjs\ Framework.framework/Helpers/nwjs\ Helper\ \(Renderer\).app
```

VScode is suffering from a very similar issue. Someone posted the above (for Vscode though) workaround fixing the issue.
https://github.com/microsoft/vscode/issues/105446

McGiverGim, 2020-11-18:
If all of you can confirm this workaround, we can add it to the Mac OS notes while we find a better fix.

ljholland, 2020-11-18:
@McGiverGim - confirmed. I'm not seeing any more crashing after running that command.

McGiverGim, 2020-11-18:
If it matters, what I see in internet is that some parts of our nw framework have it's own signature, that is invalid. The above command removes them.

The perfect solution will be replace this signature by our own, but at this moment we don't have a certificate for that.

mikeller, 2020-11-22:
@McGiverGim: I guess another way to work around this will be to remove the signature on NW.js as part of the build process.
Still no appetite to pay Apple money for their protection racket - if they start giving out free certificates for community driven non-profit open source projects like Betaflight I am all on board.

McGiverGim, 2020-11-22:
Good idea. I don't have a Mac to test it.
I suppose we need to run the `codesign` command before building it and test if it works. If it works we can add it to the pipeline or to the gulpfile.

sael17, 2020-11-23:
After having upgraded to macOS Big Sur, my VS Code C++ compiler stopped working. I can not run any of my recent or old projects and files. The problem occurs every time I try to Run Build Task or just Run.
macOS Big Sur 11.0.1 in MacBook Pro(2020)

I attached some images for easier comprehension, sorry if I didn't explain it well, I'm just a beginner :D

[image]
[image]
[image]

AndreaCCIE, 2020-11-23:
> There's a possible workaround as I was not able to reproduce the issue anymore after running the following command
> 
> ```
> codesign --remove-signature /Applications/Betaflight\ Configurator.app/Contents/Frameworks/nwjs\ Framework.framework/Helpers/nwjs\ Helper\ \(Renderer\).app
> ```
> 
> VScode is suffering from a very similar issue. Someone posted the above (for Vscode though) workaround fixing the issue.
> [microsoft/vscode#105446](https://github.com/microsoft/vscode/issues/105446)

I still have random crashes after this command.

AndreaCCIE, 2020-11-23:
Actually, nwjs still runs even after the configurator crashes or gets closed:

[image]

mikeller, 2020-11-23:
@sael17: This is an entirely unrelated problem, and not pertaining to configurator at all. Please open a new issue at https://github.com/betaflight/betaflight/issues.

Autocrit, 2020-11-24:
Betaflight Configurator is also unstable on an M1 MacBook Air (it can crash on opening, closing or when running, but is still usable some of the time). As it's running Big Sur it could be the same problem and could be NW.js related. Kiss GUI has also crashed with the report showing nwjs.

mikeller, 2020-11-28:
@AndreaCCIE: Looks like code signing is a red herring - the NW.js apps that we are using are not signed in the first place:

```
mikeller@iMac betaflight-configurator % git checkout 10.7.0
Note: switching to '10.7.0'.
mikeller@iMac betaflight-configurator % rm -r apps
mikeller@iMac betaflight-configurator % yarn gulp apps
yarn run v1.22.5
$ gulp apps
[...]
[12:47:13] Finished 'apps' after 57 s
✨  Done in 59.11s.
mikeller@iMac betaflight-configurator % for file in ./apps/betaflight-configurator/osx64/betaflight-configurator.app/Contents/Frameworks/nwjs\ Framework.framework/Helpers/*.app; do codesign -d ${file}; done
./apps/betaflight-configurator/osx64/betaflight-configurator.app/Contents/Frameworks/nwjs Framework.framework/Helpers/nwjs Helper (GPU).app: code object is not signed at all
./apps/betaflight-configurator/osx64/betaflight-configurator.app/Contents/Frameworks/nwjs Framework.framework/Helpers/nwjs Helper (Plugin).app: code object is not signed at all
./apps/betaflight-configurator/osx64/betaflight-configurator.app/Contents/Frameworks/nwjs Framework.framework/Helpers/nwjs Helper (Renderer).app: code object is not signed at all
./apps/betaflight-configurator/osx64/betaflight-configurator.app/Contents/Frameworks/nwjs Framework.framework/Helpers/nwjs Helper.app: code object is not signed at all
mikeller@iMac betaflight-configurator % 
```

Autocrit, 2020-11-28:
Rebuilding with nw.js version 0.50.1 and node.js 15.3.0, as well as doing.a codesign --deep --force --remove-signature on betaflight-configurator.app for good measure, the app seems to open, PID sliders work, and it closes without crashing so far. Big Sur with M1.

TahoeEd, 2020-12-01:
MacBook Air 2019 Big Sur 11.0.1  I am having similar issues however when going into the Setup tab I get this graphic and the program locks up.
[image]

9dc, 2020-12-14:
Even after doing what @Autocrit said, i still get random crashes :/

SPCORRIV, 2020-12-19:
MacBook Pro (Retina, 13-inch, Early 2015)
Big Sur
Issues.
1. Disconnects and reconnects randomly
2. Screens freeze cant disconnect.
3. Screen freezes random images on setup page
Thanks

SebastianKumor, 2020-12-25:
Hello, I get this error when running configurator.
[image]

mikeller, 2020-12-25:
@SebastianKumo: This is normal if you used a development version of configurator, and then switched back to the stable version running on an older version of NW.js. Delete your NW.js application settings at `~/Library/Application Support/betaflight-configurator` and you'll be fine.

SebastianKumor, 2020-12-26:
@mikeller the newest master with updated NW.js works for me on macOS big sur without crashes or any other glitches I experienced on 10.7.0

mikeller, 2020-12-26:
Thanks @SebastianKumor for the feedback!

SebastianKumor, 2020-12-26:
@mikeller would be nice if someone with M1 chip mac could test it as well. I am running on intel

9dc, 2020-12-27:
Thanks! Its working perfectly again.

kenthinson, 2021-01-03:
Hello @mikeller  is the new version with nwjs update available as a built binary or do I need to build it? Thanks! Right now on my M1 Mac I get crashes and nwjs is using 200% CPU in a left over zombie process  after the crash. I assume the merge you did is related.

Asizon, 2021-01-03:
@kenthinson you can download binaries here:
https://github.com/betaflight/betaflight-configurator-nightlies/releases

Use only for testing

kenthinson, 2021-01-03:
@Asizon Thanks it's been working great for a few hours now! :)

AndreaCCIE, 2021-01-05:
> how to deal with this issue? 

1) read the post above
2) install the 10.8 beta from https://github.com/betaflight/betaflight-configurator-nightlies/releases

TahoeEd, 2021-01-05:
> Hello friends .. I'm having the same issue .. and i'm new I don't know what to do .. Betaflight 10.7 on Big Sur (MacBook Air 2020)
>
> how to deal with this issue? please help .. I have too many crashes .. I can't setup my quads !!
>
> thanks

Mine does the same.  The guys here are making progress.  You just have to be patient.  I have been able to set my quads up but it may take several restarts to do it.

Autocrit, 2021-01-06:
> You just have to be patient.

I hope you won't mind if I disagree with waiting for the next release when there are nightly builds with the necessary NW.js version update.

You could even build 10.7.0 with NW.js v0.50.2 if you were concerned about changes since that point.

Nalkhashab, 2021-01-06:
> > You just have to be patient.
> 
> I hope you won't mind if I disagree with waiting for the next release when there are nightly builds with the necessary NW.js version update.
> 
> You could even build 10.7.0 with NW.js v0.50.2 if you were concerned about changes since that point.

how to do this? i mean NW.js ??

AndreaCCIE, 2021-01-06:
As already written:
1. read the posts above
2. install the 10.8 beta from https://github.com/betaflight/betaflight-configurator-nightlies/releases

Autocrit, 2021-01-06:
Yup, as AndreaCCIE says. If you're interested in the source, the NW.js version is specified in gulpfile.js and instructions on how to build Betaflight Configurator are in the README.md.

SebastianKumor, 2021-01-07:
@Nalkhashab just pull the master branch and build it following the guide in the read me

MattBol84, 2021-03-22:
Hello, 
I use MacBook Pro 16 inch, 2,6 GHz Intel Core i7 6 core, late 2019. 
Big sur upgraded one week ago... Betaflight is totally unusable. I saw all the issues reported here.

ljholland, 2021-03-22:
> Hello,
> I use MacBook Pro 16 inch, 2,6 GHz Intel Core i7 6 core, late 2019.
> Big sur upgraded one week ago... Betaflight is totally unusable. I saw all the issues reported here.

While it's completely annoying, restarting the app several times when it crashes, seems to get it working. Don't give up. Sometimes, I'll get into the configuration tab, and then I'll switch tabs and then it'll crash...but I'll keep restarting and can eventually make the changes needed.  It's a major pain in the a$$.

AndreaCCIE, 2021-03-22:
> Hello,
> I use MacBook Pro 16 inch, 2,6 GHz Intel Core i7 6 core, late 2019.
> Big sur upgraded one week ago... Betaflight is totally unusable. I saw all the issues reported here.

There are not only the issues, even the solutions.

SebastianKumor, 2021-03-22:
@MattBol84 You can build yourself new version of betaflight configuratior from master branch. If you do not know how I can upload the build I am using on google drive. It works flawlessly on big sur

ljholland, 2021-03-22:
> @MattBol84 You can build yourself new version of betaflight configuratior from master branch. If you do not know how I can upload the build I am using on google drive. It works flawlessly on big sur

This would be great!

mikeller, 2021-03-22:
@ljholland:

> If you do not know how I can upload the build I am using on google drive. It works flawlessly on big sur

You could also just use a nightly build if you want to use an unstable development version:

https://github.com/betaflight/betaflight-configurator-nightlies/releases

MattBol84, 2021-03-22:
@SebastianKumor Unfortunately, I'm not able to build by myself.. Can I download your one from Google drive ? 
It would be great!

MattBol84, 2021-03-22:
@mikeller I already tried to install that version, but with no results :(

SebastianKumor, 2021-03-22:
@MattBol84 for me also the nightly build works but here try this [drive](https://drive.google.com/drive/folders/1_9RXjhbTAW6zVxuYLPeHN0TXbWYFlE6Q?usp=sharing) link. I am running on latest big sur on intel i9 mac book pro and it works without any problems. Flashed 2 quads with it and everything is fine

mikeller, 2021-03-22:
@MattBol84: Can you please provide a couple more details than 'with no results' - if there is something wrong with the nightly builds then this will have to be fixed, but without a description of the problem this won't be possible.
Also, I would be careful about installing things that other users offer for download on the internet and that claim to be identical to the official build - there really isn't much reason for a user to do this except if they want to infect your computer with malware...

SebastianKumor, 2021-03-22:
@MattBol84 yes, best you can do is build it yourself.

ljholland, 2021-03-22:
The nightly build does seem more stable than the version had installed. Bad on me for not keeping up to date

mikeller, 2021-03-23:
@ljholland: No surprise there. The latest released version (10.7.0) is older than Big Sur. :wink:

MattBol84, 2021-03-23:
@mikeller Hi, sorry about waiting for my answer.  So, I've Installed 10.8, surely it's more stable and I didn't saw crashes or bugs about graphics elements. I have many problems with the VTX panel, when I try to change the power it changes on the screen but not on VTX, same for band and channel selection, then I have a problem with the "pit mode" cause randomly it works when is off or it doesn't work when is on, not always but often. One more thing: that version  (10.8) doesn't recognize the VTX smart audio. I'm using the same vtx as always, and I never had this problem with the previous OS. 
Should be everything.. hope it's all clear, my English it's terrible.

MattBol84, 2021-03-23:
@SPCORRIV The vtx as the vtx table is ok, usually when I made a change on power or channel from betaflight immediately after "Save" it changes his value.  I can check this because the led colour's different  and  I need to shift to the new channel to see the video

artem-alexandrov, 2021-05-01:
Is it possible to get 10.7.1 release version with this fix?

SebastianKumor, 2021-05-02:
@artem-alexandrov you can use the nightly build https://github.com/betaflight/betaflight-configurator-nightlies/releases . It has the fix

barts83eu, 2021-05-02:
M1 Mac mini 2020, also crashes betaflight co figurative tried 10.70 and 10.80 form April 2021, 
crashes macos and restarts 

Drone: iflight protek35

petertorelli, 2021-05-17:
Can we get some clarity on the flight controller's MCU in the crashes? I have run into issues with STM and MacOS using several JavaScript libraries for serial ports. My go-to is npm's serialport, but I've seen the port disconnect with lots of tx/rx across the ST-Link VCP (even at low baud). I'm wondering if (a) the failures in question happen on STM hardware, and (b) if that hardware is utilizing the ST-Link VCP and not a dedicated UART?

haslinghuis, 2021-05-17:
F405 and F411 are dominating F722 and F745 following Google Analytics. Afaik most boards nowadays are using VCP instead of Silabs CP2xx, we have had 2,510,000,197,078 flashes in the last year.

petertorelli, 2021-05-17:
@haslinghuis 2.5 trillion flashes? Wow. :) I have a sneaking suspicion this is a combination of changes to mac VCP drivers & STM stlink. This is something I'm working on as a side project (serial port stress test via Node/ElectronJS), but don't have a definitive date.

haslinghuis, 2021-05-17:
Well in development I flash multiple times within an hour, this is not shown in analytics. Also should compare on Windows and Linux. Also want to mention https://web.dev/serial/

haslinghuis, 2021-05-17:
Well in development I flash multiple times within an hour. Should compare OSX drivers and behavior with Windows and Linux. Also want to mention https://web.dev/serial/

[19 more replies]

closed 2022-01-19 by haslinghuis