betaflight/betaflight-configurator #2275: problems on linux

ulph0, 2020-11-12:
hi,

i am runnig the configurator 10.7.0 on linux, ubuntu 20.04 LTS. i deinstalled modemmanager, libatomic1 is installed.

every time i click on "save and reboot" in the configuration tab, the serial connection is broken afterwards, the serial device doesn't reappear, i cannot reconnect. also, the whole network hangs, which means, i don't have no more internet connection.

the same happens when i am in CLI and switch to another tab without saving. but only then. i i save, it reboots and reconnects fine.

regards.

haslinghuis, 2020-11-12:
Did you add yourself to the dialout group and login again:

    $ sudo usermod -a -G dialout $USER

ulph0, 2020-11-12:
yes. did that in the beginning some weeks ago. just checked it, i am still in dialout.

haslinghuis, 2020-11-12:
Looks like configurator still thinks it's connected. What happens when you replug the usb cable?

ulph0, 2020-11-12:
not much. network is still frozen, bc says can't connect to serial port.

haslinghuis, 2020-11-12:
Can you copy paste the log? Try another usb data cable.
```
2020-11-12 @ 21:21:22 -- Running - OS: Linux, Chrome: 80.0.3987.116, Configurator: 10.7.0
2020-11-12 @ 21:21:23 -- Loaded release information for configurator from GitHub.
2020-11-12 @ 21:21:36 -- Serial port successfully opened with ID: 1
2020-11-12 @ 21:21:36 -- MultiWii API version: 1.43.0
2020-11-12 @ 21:21:36 -- Flight controller info, identifier: BTFL, version: 4.2.4
2020-11-12 @ 21:21:36 -- Running firmware released on: Oct 20 2020 08:19:06
2020-11-12 @ 21:21:36 -- Board: MTKS/MATEKF411(STM32F411), version: 0
2020-11-12 @ 21:21:36 -- Unique device ID: 0x5600663337511838373732
2020-11-12 @ 21:21:36 -- Craft name: MF411
2020-11-12 @ 21:21:36 -- Arming Disabled
2020-11-12 @ 21:21:40 -- EEPROM saved
2020-11-12 @ 21:21:40 -- Device - Rebooting
2020-11-12 @ 21:21:40 -- Serial port successfully closed
2020-11-12 @ 21:21:43 -- Serial port successfully opened with ID: 2
2020-11-12 @ 21:21:43 -- MultiWii API version: 1.43.0
2020-11-12 @ 21:21:43 -- Flight controller info, identifier: BTFL, version: 4.2.4
2020-11-12 @ 21:21:43 -- Running firmware released on: Oct 20 2020 08:19:06
2020-11-12 @ 21:21:43 -- Board: MTKS/MATEKF411(STM32F411), version: 0
2020-11-12 @ 21:21:43 -- Unique device ID: 0x5600663337511838373732
2020-11-12 @ 21:21:43 -- Craft name: MF411
2020-11-12 @ 21:21:43 -- Arming Disabled
2020-11-12 @ 21:21:50 -- EEPROM saved
2020-11-12 @ 21:21:50 -- Device - Rebooting
2020-11-12 @ 21:21:50 -- Serial port successfully closed
2020-11-12 @ 21:21:53 -- Serial port successfully opened with ID: 3
...
```

ulph0, 2020-11-12:
which log? where does it resist?

haslinghuis, 2020-11-12:
[image]

haslinghuis, 2020-11-12:
Select the text, copy and paste here please.

ulph0, 2020-11-12:
ok got it. not so easy, i had to reboot the computer to get internet back (even restarting the network services doesn't work) this is all what it says. is there a debug mode?

2020-11-12 @ 21:36:12 -- Running - OS: Linux, Chrome: 80.0.3987.116, Configurator: 10.7.0
2020-11-12 @ 21:36:12 -- Using cached release information for configurator releases.
2020-11-12 @ 21:36:25 -- Serial port successfully opened with ID: 1
2020-11-12 @ 21:36:25 -- MultiWii API version: 1.40.0
2020-11-12 @ 21:36:25 -- Flight controller info, identifier: BTFL, version: 3.5.6
2020-11-12 @ 21:36:25 -- Running firmware released on: Feb 16 2019 13:28:02
2020-11-12 @ 21:36:25 -- Board: OMNIBUS, version: 0
2020-11-12 @ 21:36:25 -- Unique device ID: 0x1800215634570320353233
2020-11-12 @ 21:36:25 -- Craft name: gaga
2020-11-12 @ 21:36:25 -- Arming Disabled
2020-11-12 @ 21:36:35 -- EEPROM saved
2020-11-12 @ 21:36:35 -- Device - Rebooting
2020-11-12 @ 21:36:36 -- Serial port successfully closed
2020-11-12 @ 21:36:49 -- Failed to open serial port
2020-11-12 @ 21:36:59 -- Failed to open serial port

haslinghuis, 2020-11-12:
Debug is not enabled in release versions afaik. Using Ubuntu 20.04 on this laptop for months and never had network issues while using configurator. Try removing the cache first and test if this helps:

    $ rm -rf ~/.cache/betaflight-configurator

Nexy try to reinstall the configurator
Otherwise reinstall Ubuntu :wink:

ulph0, 2020-11-12:
will try that one with the cache, reinstall bc, trying a third cable. won't reinstall ubuntu, runs pretty well, also no other software on this system has problems with serial communication over usb.

haslinghuis, 2020-11-12:
As it worked before something has changed - so really curious about any conflict.

ulph0, 2020-11-12:
i tried all that, also on different usb ports ... nothing helped. which doesn't takes me wonder since communicating over the port seems not to be the problem. the problem appears when betaflight configurator closes the connection to that port.

haslinghuis, 2020-11-12:
Please show the output of

    $ ls -t /var/lib/dpkg/info/*.list | sed -e 's/\.list$//' | head -n 50

ulph0, 2020-11-12:
/var/lib/dpkg/info/betaflight-configurator
/var/lib/dpkg/info/librxtx-java
/var/lib/dpkg/info/picocom
/var/lib/dpkg/info/slack-desktop
/var/lib/dpkg/info/openjfx-source
/var/lib/dpkg/info/openjfx
/var/lib/dpkg/info/openjdk-8-jre-headless:amd64
/var/lib/dpkg/info/openjdk-8-jre:amd64
/var/lib/dpkg/info/openjdk-8-jdk-headless:amd64
/var/lib/dpkg/info/openjdk-8-jdk:amd64
/var/lib/dpkg/info/openjdk-11-jre-headless:amd64
/var/lib/dpkg/info/openjdk-11-jre:amd64
/var/lib/dpkg/info/network-manager-config-connectivity-ubuntu
/var/lib/dpkg/info/libopenjfx-java
/var/lib/dpkg/info/libopenjfx-jni
/var/lib/dpkg/info/libefiboot1:amd64
/var/lib/dpkg/info/libefivar1:amd64
/var/lib/dpkg/info/gnome-initial-setup
/var/lib/dpkg/info/gir1.2-nma-1.0:amd64
/var/lib/dpkg/info/libnma0:amd64
/var/lib/dpkg/info/network-manager-gnome
/var/lib/dpkg/info/gir1.2-nm-1.0:amd64
/var/lib/dpkg/info/libnm0:amd64
/var/lib/dpkg/info/network-manager
/var/lib/dpkg/info/apport-gtk
/var/lib/dpkg/info/apport
/var/lib/dpkg/info/python3-apport
/var/lib/dpkg/info/python3-problem-report
/var/lib/dpkg/info/acpi-call-dkms
/var/lib/dpkg/info/plymouth
/var/lib/dpkg/info/plymouth-label
/var/lib/dpkg/info/plymouth-theme-ubuntu-logo
/var/lib/dpkg/info/plymouth-theme-spinner
/var/lib/dpkg/info/libplymouth5:amd64
/var/lib/dpkg/info/libmaxminddb0:amd64
/var/lib/dpkg/info/intel-microcode
/var/lib/dpkg/info/libraptor2-0:amd64
/var/lib/dpkg/info/linux-modules-5.4.0-51-generic
/var/lib/dpkg/info/linux-image-5.4.0-51-generic
/var/lib/dpkg/info/linux-modules-extra-5.4.0-51-generic
/var/lib/dpkg/info/linux-tools-virtual
/var/lib/dpkg/info/linux-tools-5.4.0-53-generic
/var/lib/dpkg/info/linux-tools-5.4.0-53
/var/lib/dpkg/info/linux-tools-common
/var/lib/dpkg/info/linux-libc-dev:amd64
/var/lib/dpkg/info/linux-image-generic-hwe-18.04
/var/lib/dpkg/info/linux-headers-generic-hwe-18.04
/var/lib/dpkg/info/linux-generic-hwe-18.04
/var/lib/dpkg/info/linux-headers-generic
/var/lib/dpkg/info/linux-headers-5.4.0-53-generic

haslinghuis, 2020-11-12:
Still strange because you mentioned it worked for weeks.
Nothing obvious - it's the order in what's installed on your system - cannot compare now because I moved on to WSL2 (and graphics support is almost implemented - don't want to use the current hack) and my other laptop / pc are running Fedora.
To test you could build configurator from source yourself and use F12 to see what error pops up.
https://github.com/betaflight/betaflight-configurator#development

ulph0, 2020-11-13:
that is a misunderstanding. i have these problems from the beginning on. what is WSL2? here is the dmesg output, i hope without compromising data:

[   81.001648] show_signal_msg: 42 callbacks suppressed
[   81.001657] appimagelaunche[2250]: segfault at 200010028 ip 00007f4e57694704 sp 00007ffca1ba02d0 error 4 in libQt5Core.so.5.12.8[7f4e57567000+2e0000]
[   81.001678] Code: 41 54 55 53 48 83 ec 38 4c 8b 2f 48 8b 1e 64 48 8b 04 25 28 00 00 00 48 89 44 24 28 31 c0 49 39 dd 0f 84 17 01 00 00 49 89 fc <49> 8b 7d 28 48 8b 43 28 48 89 f5 48 85 ff 74 6c 48 85 c0 74 3f 48
[  121.018203] traps: appimagelaunche[3024] general protection fault ip:7f743eb82704 sp:7ffc4fc84060 error:0 in libQt5Core.so.5.12.8[7f743ea55000+2e0000]
[  149.975765] ------------[ cut here ]------------
[  149.975781] WARNING: CPU: 2 PID: 0 at kernel/workqueue.c:1477 __queue_work+0x348/0x3f0
[  149.975782] Modules linked in: acpi_call(OE) vboxnetadp(OE) vboxnetflt(OE) vboxdrv(OE) rfcomm cmac algif_hash algif_skcipher af_alg bnep ccm btusb btrtl btbcm btintel bluetooth ecdh_generic ecc cdc_acm intel_rapl_msr mei_hdcp intel_rapl_common x86_pkg_temp_thermal intel_powerclamp snd_hda_codec_hdmi coretemp kvm_intel kvm snd_hda_codec_realtek snd_hda_codec_generic crct10dif_pclmul snd_hda_intel iwlmvm snd_seq_midi snd_intel_dspcfg ghash_clmulni_intel mac80211 snd_seq_midi_event i915 libarc4 snd_rawmidi iwlwifi snd_hda_codec aesni_intel snd_hda_core crypto_simd snd_hwdep cryptd glue_helper rapl snd_seq intel_cstate joydev snd_pcm input_leds serio_raw cfg80211 thinkpad_acpi rtsx_pci_ms wmi_bmof nvram drm_kms_helper snd_seq_device memstick i2c_algo_bit ledtrig_audio mei_me fb_sys_fops mei snd_timer syscopyarea snd sysfillrect sysimgblt soundcore intel_smartconnect mac_hid sch_fq_codel binfmt_misc parport_pc ppdev lp parport drm ip_tables x_tables autofs4 hid_generic rtsx_pci_sdmmc ahci
[  149.975876]  psmouse libahci usbhid crc32_pclmul i2c_i801 lpc_ich rtsx_pci e1000e hid wmi video
[  149.975895] CPU: 2 PID: 0 Comm: swapper/2 Tainted: G           OE     5.4.0-53-generic #59-Ubuntu
[  149.975897] Hardware name: XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
[  149.975904] RIP: 0010:__queue_work+0x348/0x3f0
[  149.975909] Code: 59 75 76 a9 00 01 1f 00 75 0f 65 48 8b 3c 25 c0 6b 01 00 f6 47 24 20 75 25 0f 0b 48 83 c4 18 5b 41 5c 41 5d 41 5e 41 5f 5d c3 <0f> 0b e9 fd fd ff ff 41 83 cd 02 48 8d 53 60 e9 e2 fd ff ff e8 3f
[  149.975912] RSP: 0018:ffffa3ed80164c40 EFLAGS: 00010002
[  149.975916] RAX: ffff93e44aa78790 RBX: ffff93e4564af700 RCX: 0000000000000000
[  149.975918] RDX: ffff93e44aa78798 RSI: 0000000000000004 RDI: ffff93e4564aa680
[  149.975921] RBP: ffffa3ed80164c80 R08: ffff93e455002910 R09: ffff93e455002958
[  149.975924] R10: 0000000000000000 R11: ffffffff8ae5e4b8 R12: 0000000000000014
[  149.975926] R13: ffff93e45540ce00 R14: 0000000000002000 R15: ffff93e44aa78790
[  149.975930] FS:  0000000000000000(0000) GS:ffff93e456480000(0000) knlGS:0000000000000000
[  149.975932] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  149.975935] CR2: 00007f7c7051d000 CR3: 000000018d20a002 CR4: 00000000001606e0
[  149.975938] Call Trace:
[  149.975941]  <IRQ>
[  149.975950]  queue_work_on+0x3b/0x50
[  149.975962]  acm_write_bulk+0x70/0x80 [cdc_acm]
[  149.975968]  __usb_hcd_giveback_urb+0x7a/0x120
[  149.975972]  usb_hcd_giveback_urb+0xca/0xe0
[  149.975980]  xhci_giveback_urb_in_irq.isra.0+0x7b/0xf0
[  149.975985]  xhci_td_cleanup+0xf6/0x140
[  149.975990]  finish_td+0xcc/0x190
[  149.975997]  handle_tx_event+0x55d/0x11c0
[  149.976004]  ? raw_notifier_call_chain+0x16/0x20
[  149.976011]  xhci_irq+0x27a/0x3d0
[  149.976017]  xhci_msi_irq+0x11/0x13
[  149.976024]  __handle_irq_event_percpu+0x42/0x180
[  149.976030]  handle_irq_event_percpu+0x33/0x80
[  149.976036]  handle_irq_event+0x3b/0x5a
[  149.976041]  handle_edge_irq+0x93/0x1c0
[  149.976046]  do_IRQ+0x55/0xf0
[  149.976053]  common_interrupt+0xf/0xf
[  149.976055]  </IRQ>
[  149.976062] RIP: 0010:cpuidle_enter_state+0xc5/0x450
[  149.976067] Code: ff e8 bf cc 80 ff 80 7d c7 00 74 17 9c 58 0f 1f 44 00 00 f6 c4 02 0f 85 65 03 00 00 31 ff e8 82 36 87 ff fb 66 0f 1f 44 00 00 <45> 85 ed 0f 88 8f 02 00 00 49 63 cd 4c 8b 7d d0 4c 2b 7d c8 48 8d
[  149.976069] RSP: 0018:ffffa3ed800cbe38 EFLAGS: 00000246 ORIG_RAX: ffffffffffffffdd
[  149.976074] RAX: ffff93e4564aad00 RBX: ffffffff8af59c20 RCX: 000000000000001f
[  149.976076] RDX: 0000000000000000 RSI: 0000000033518adf RDI: 0000000000000000
[  149.976079] RBP: ffffa3ed800cbe78 R08: 00000022eb3ffd91 R09: 0000000000000001
[  149.976081] R10: ffff93e4564a9a00 R11: ffff93e4564a99e0 R12: ffff93e4564b5300
[  149.976084] R13: 0000000000000005 R14: 0000000000000005 R15: ffff93e4564b5300
[  149.976091]  ? cpuidle_enter_state+0xa1/0x450
[  149.976096]  cpuidle_enter+0x2e/0x40
[  149.976101]  call_cpuidle+0x23/0x40
[  149.976104]  do_idle+0x1dd/0x270
[  149.976108]  cpu_startup_entry+0x20/0x30
[  149.976113]  start_secondary+0x167/0x1c0
[  149.976119]  secondary_startup_64+0xa4/0xb0
[  149.976123] ---[ end trace 24a33aa7c2c3721f ]---
[  150.195828] usb 2-2: USB disconnect, device number 3
[  150.195975] cdc_acm 2-2:1.0: failed to set dtr/rts
[  161.033475] traps: appimagelaunche[5099] general protection fault ip:7f7706101704 sp:7ffe889be4d0 error:0 in libQt5Core.so.5.12.8[7f7705fd4000+2e0000]

haslinghuis, 2020-11-13:
It should look like this:
```
[16727.300837] usb 3-4: USB disconnect, device number 17
[16731.081805] usb 3-4: new full-speed USB device number 18 using xhci_hcd
[16731.209478] usb 3-4: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 2.00
[16731.209485] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[16731.209488] usb 3-4: Product: Betaflight STM32F411
[16731.209492] usb 3-4: Manufacturer: Betaflight
[16731.209495] usb 3-4: SerialNumber: 0x8000000
[16731.211417] cdc_acm 3-4:1.0: ttyACM0: USB ACM device
[16738.336682] usb 3-4: USB disconnect, device number 18
[16738.336795] cdc_acm 3-4:1.0: failed to set dtr/rts
[16740.801992] usb 3-4: new full-speed USB device number 19 using xhci_hcd
[16740.929354] usb 3-4: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 2.00
[16740.929361] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[16740.929365] usb 3-4: Product: Betaflight STM32F411
[16740.929368] usb 3-4: Manufacturer: Betaflight
[16740.929371] usb 3-4: SerialNumber: 0x8000000
[16740.930504] cdc_acm 3-4:1.0: ttyACM0: USB ACM device
```

ulph0, 2020-11-13:
i found a post where somebody says "depending on the flightcontroller it is possible that you have to customize the udev rules". do you know what he means?

haslinghuis, 2020-11-13:
Please take a look at: https://github.com/betaflight/betaflight/blob/master/docs/development/Building%20in%20Ubuntu.md#flashing-a-target-with-betaflight-configurator-on-ubuntu-2004

ulph0, 2020-11-13:
doesn't help :( will give up for now. good night and thank you.

haslinghuis, 2020-11-13:
I have just clean installed Ubuntu 20.04 on a pc and it all just works. Only had to remove `modemmanager` and put myself in `dialout` group. After that I can connect and reboot without any problems.

ulph0, 2020-11-15:
meanwhile i also had upgraded another system to ubuntu 20.04 where bc runs normally. this doesn't really help, because i need it to run on my laptop. so to describe it again: i have only problems when the connection to the serial port is losed by configuration tab and if i am changing from CLI tab to another without saving. everything else, first config-writing, flashing with following reboot, CLI with save is working. if somebody has an idea what could cause this, please contact me, i will close this ticket for now.

closed 2020-11-15 by ulph0

mikeller, 2020-11-22:
@ulph0: Idea: yes. Helpful: sadly no. I've seen similar kinds of misbehaviour (kernel throwing stack traces) in linux when trying to use USB 3 chipsets that were not fully supported under linux. In the case of a laptop not sure there is much that can be done to address this - maybe check out all of the USB ports / try puttting a USB hub between flight controller and laptop to see if any combination works.

ulph0, 2021-01-30:
thanks mikeller meanwhile i also upgraded from thinkpad t440 to x260. new cable, new usb hub, also without hub, same problem. either it's a problem with ubuntu on thinkpads or a problem with my ubuntu. and it does only appears when i am connected to betaflight via usb.

mikeller, 2021-02-01:
@ulph0: That's weird - I am using Ubuntu on a Thinkpad T460, and connecting to flight controllers is working just fine - but potentially that's an entirely different chipset to the two laptops that you are using.