betaflight/betaflight-configurator #2278: Help required

MagentoAmator, 2020-11-16:
Wanted to play with BF configurator a little but while compiling throws me an error:
Would anyone help me to locate the issue please?
$ gulp release --win32
[ '/root/.nvm/versions/node/v10.23.0/bin/node',
  '/var/www/html/purefpv.co.uk/public_html/betaflight/node_modules/.bin/gulp',
  'release',
  '--win32' ]
Building for platform(s): win32.
[22:30:54] Using gulpfile /var/www/html/purefpv.co.uk/public_html/betaflight/gulpfile.js
[22:30:54] Starting 'release'...
[22:30:54] Starting 'clean_release'...
[22:30:54] Starting 'clean_apps'...
[22:30:54] Starting 'clean_dist'...
[22:30:54] Finished 'clean_release' after 50 ms
[22:30:54] Finished 'clean_apps' after 251 ms
[22:30:57] Finished 'clean_dist' after 3.3 s
[22:30:57] Starting 'dist_src'...
[22:30:58] Finished 'dist_src' after 1.24 s
[22:30:58] Starting 'dist_changelog'...
[22:30:58] Finished 'dist_changelog' after 11 ms
[22:30:58] Starting 'dist_yarn'...
[1/5] Validating package.json...
[2/5] Resolving packages...
[3/5] Fetching packages...
info fsevents@1.2.9: The platform "linux" is incompatible with this module.
info "fsevents@1.2.9" is an optional dependency and failed compatibility check. Excluding it from installation.
info fsevents@2.1.1: The platform "linux" is incompatible with this module.
info "fsevents@2.1.1" is an optional dependency and failed compatibility check. Excluding it from installation.
info fsevents@2.1.3: The platform "linux" is incompatible with this module.
info "fsevents@2.1.3" is an optional dependency and failed compatibility check. Excluding it from installation.
[4/5] Linking dependencies...
[5/5] Building fresh packages...
success Saved lockfile.
[22:31:16] Finished 'dist_yarn' after 18 s
[22:31:16] Starting 'dist_locale'...
[22:31:16] Finished 'dist_locale' after 234 ms
[22:31:16] Starting 'dist_libraries'...
[22:31:16] Finished 'dist_libraries' after 121 ms
[22:31:16] Starting 'dist_resources'...
[22:31:17] Finished 'dist_resources' after 423 ms
[22:31:17] Starting 'dist_rollup'...
postcss.plugin was deprecated. Migration guide:
https://evilmartians.com/chronicles/postcss-8-plugin-migration
[22:31:25] Finished 'dist_rollup' after 8.08 s
[22:31:25] Starting 'getHash'...
[22:31:25] Finished 'getHash' after 48 ms
[22:31:25] Starting 'writeChangesetId'...
[22:31:25] Finished 'writeChangesetId' after 10 ms
[22:31:25] Starting 'cordova_dist_none'...
[22:31:25] Finished 'cordova_dist_none' after 909 μs
[22:31:25] Starting 'apps'...
Using v0.47.0 (normal)
Create cache folder in /var/www/html/purefpv.co.uk/public_html/betaflight/cache/0.47.0-normal
Using cache for: win32
Create release folder in /var/www/html/purefpv.co.uk/public_html/betaflight/apps/betaflight-configurator/win32
Error building NW apps: Error: spawn wine ENOENT
[22:31:28] The following tasks did not complete: release, <parallel>, <series>, apps
[22:31:28] Did you forget to signal async completion?
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.

Installed every dependencies but still can't make release file properly.

Regards
Robiniosa

mikeller, 2020-11-16:
@Robiniosa: Cross-compiling the Windows version of configurator on linux requires wine and a bunch of mono and Microsoft .net libraries installed. It is therefore not recommended. Build for the architecture you are working on.

Robiniosa, 2020-11-17:
Hello there,
Thank you for the hint. So what is the best configuration to avoid those issues ? Is there any guide which i could follow ? I have no problem with linux so i can try on it anyway.  Just need advice what else is missing last night installed some wine packages as well but it still does not work. 

Regards 
Robin

From my iPhone

mikeller, 2020-11-17:
@Robiniosa: https://github.com/betaflight/betaflight-configurator#app-build-via-nwjs-windowslinuxmacos-or-cordova-android

closed 2021-01-08 by github-actions[bot]