betaflight/betaflight-configurator #2279: [FEATURE REQUEST] Failsafe : "LAND" after failed sanity checks

2N5R, 2020-11-17:
I hope i'm right here.

Wouldn't it be a good idea to add an option that prevents a quadcopter from disarming (falling out of the sky) in case of a failsafe event with sanity checks failed? An option like "land in case of sanity check failed" would make sense. 

Happend today: Failsafe event, GPS lost right in that moment, quad dropped on the concrete from 250ft height. A mode "land in case of sanity check failed" would have prevented the destruction.

Again, I hope I am right here!

Thanks and greetings

McGiverGim, 2020-11-18:
Betaflight is divided into several parts: the firmware is what goes into the FC, and the Configurator is what serves to configure it. To configure it, before we need that the FC (firmware) supports it. Your feature request must be implemented before into the firmware, so you can open your feature request there: https://github.com/betaflight/betaflight

closed 2020-11-18 by McGiverGim

mikeller, 2020-11-22:
Duplicate of https://github.com/betaflight/betaflight/issues/10353.