betaflight/betaflight-configurator #2281: Button to open Betaflight

rcpilotp51, 2020-11-19:
Hey Guys! Thanks for your help in the past.

I was wondering if anyone knows how to open betaflight using either a link (href) or a javascript function.

My goal is to open the app either on windows or mac and populate the navbar with "Manual Selection" with the port: (ie 'tcp://blah.blah')

Possible or no?
I dont really know where to ask and Google is not my friend today...

haslinghuis, 2020-11-22:
I envision a cloud configurator in the future anyway using `Serial` or `USBDevice` API.

But for now `tcp` connection is only half working. It needs a serial connection after a reboot to reconnect again.

At one point `tcp` reconnect did work but some minor changes quickly led to another flow in the code which I still can't get my head around. So my `working` solution was not future proof anyway.

I didn't try to mimic the configurator in separate code however for further investigation, like trying another API.

mikeller, 2020-11-22:
@rcpilotp51: Definitely possible - the way to get this done is to define a custom protocol (`btfltcp://`?) and have the configurator installer register this with the operating system - then any app handling URLs will open Betaflight configurator when encountering a `btfltcp://...` URL.

haslinghuis, 2024-05-15:
We have dropped support for NWjs in upcoming version (11.0). Soon can use URL directly.

closed 2024-05-15 by haslinghuis