betaflight/betaflight-configurator #2301: Feature Request - Please add the displayName of the port in addition to the default path

haybeseret, 2020-12-01:
https://github.com/betaflight/betaflight-configurator/blob/a8b43e1e7c3334b09b0fc9080cc0e4104dd59a1e/src/js/port_handler.js#L17

Hello,
Now days i see more and more computer that has multiple serial connections.
It is a little bit time consuming to go to the device manager and look for that STM device and then choose it's correct COM port on the configurator. 
I think it autodetect the new STM device - but i am not sure it always works.
It will be very helpful to get actual displayname in addition to the com port (I think that blheli suits are also doing this.)
for example instead of the simple COM1,COM2,COM3 array to see: 

Standard Serial over Bluetooth link(COM7)
Standard Serial over Bluetooth link(COM8)
Intel(R) Active Management Technology - SOL(COM3)
STMicroelectronics Virtual COM Port(COM4)

Thanks for all the amazing work guys!

Asi.

McGiverGim, 2020-12-01:
This has been done some time ago. Have you tested latest nightly?

haybeseret, 2020-12-01:
> This has been done some time ago. Have you tested latest nightly?

Did not test as i was trying to look for a commit and couldn't find any.
I see that it is indeed exists in the nightly (amazing !!) .
Just need to change the #portsinput width on the main.css to something like 300 so it will look better.

[image]

Thanks again !

closed 2020-12-17 by McGiverGim