betaflight/betaflight-configurator #2304: Rearrange motor and servo tab

haslinghuis, 2020-12-04:
Feature Request as suggested by @McGiverGim:

- [ ] Move the servos test to the servos tab, to make some room here.
- [ ] Move the all motor configuration like protocol, poles, etc. from configuration tab to the motors tab.

haslinghuis, 2020-12-23:
- [x] Move ESC/Motor features from Configuration TAB to motors TAB. Configuration TAB is ready.
- [ ] MOTOR_STOP not showing
- [ ] ESC_SENSOR not showing
- [ ] 3D ESC/Motor Features not showing
- [ ] Remove mixer from Graph
- [ ] Move servo test to servos.js
- [ ] Maybe even move Dshot Beacon Configuration

Additional thoughts on Configuration TAB:
- [ ] Move System Configuration to sensor TAB
- [ ] Move board and sensor alignment / Accelerometer Trim to sensor TAB
- [ ] Move RSSI / Receiver / Arming to receiver TAB
- [ ] Move Receiver to receiver TAB
- [ ] Move Personalization / Camera to osd TAB
- [ ] Convert GPS to a feature and move GPS config to GPS TAB
- [ ] Move remaining Features into their respective TABs.

[image]

[image]

[image]

[image]

haslinghuis, 2020-12-25:
Is this an old setting setting, because when using arming on an aux channel this should be automatic.

[image]

haslinghuis, 2020-12-25:
Would be nice to have a grid layout where the user can swap panes

haslinghuis, 2020-12-26:
There seems to be a bug I didn't notice before:

[image]

limonspb, 2020-12-26:
What's wrong here?

haslinghuis, 2021-01-02:
[image]

haslinghuis, 2021-01-08:
[image]

haslinghuis, 2021-01-08:
[image]

haslinghuis, 2021-01-09:
[image]

McGiverGim, 2021-01-09:
Great work here! Is taking some time but the result is very good!

haslinghuis, 2021-01-09:
[image]

haslinghuis, 2021-01-09:
Last thing before done is to remove fc,js - it's part of another PR I will make ready asap - because this PR depends on it

mikeller, 2021-01-09:
I am still not convinced that the square 'bars' and the tiny sliders are good UX.
I think if we arranged them side by side we'd actually make the slicers usable instead of 'pro forma' only.
Alternatively, if we don't think that users will want to use the sliders, we should remove them and replace them with numerical input fields, and use the space we gain to change the 'bars' to actually be bars and not squares.

haslinghuis, 2021-01-09:
@mikeller: The minimum size would only take about 30% width at full HD. So we could save a lot of space.

If we also move the notice below the test we now have a column on the right for the configuration section.

[image]

mikeller, 2021-01-10:
I prefer something along the lines of this - having both the sliders and the bars side-by-side seems to be much more conducive to represent them in the optimal form of being tall and narrow:

[image]

haslinghuis, 2021-01-10:
Wow @Asizon , this idea came from @limonspb any ideas?

haslinghuis, 2021-01-13:
Besides webkit there also :smirk: many Vue sliders like https://github.com/xwpongithub/vue-range-slider

haslinghuis, 2021-01-13:
[image]

PR #2356 (2021-02-13): Arrange configuration and motor tab
Fixes: #2304 

**Summary**: 
- [x] Move mixer and motor/esc features from configuration to motor tab
- [x] Move servo test to servos tab

**Suggestions**:
- [x] Styling changes notice, mixer and buttons @Asizon / @limonspb 
- [x] Fix `MOTOR_STOP` alignment with `ESC_SENSOR` and align padding on different places @TheIsotopes
- [x] Move reorder Motors and Motor direction to Mixer settings @zzyzx / @limonspb
- [x] Move mixer from graph to configuration and put configuration before testing @McGiverGim  
- [x] Stop motor testing when making configuration changes  [... cut]
closed 2021-02-13 by mikeller