betaflight/betaflight-configurator #2330: OSD Preview is not rendered

haslinghuis, 2020-12-09:
Found out that with latest firmware development build the image in configurator does show only the top bar of the image.

Flashed back to latest release. Turns out after enabling all OSD elements I'll get the same. Reducing elements suddenly shows the OSD picture in configurator. Disabling `Stick overlay right` does show the OSD screen in configurator again. But sometimes you have to disable `Stick overlay left` or `both`. Sometimes have to save and switch tabs before it updates, sometimes only clicking the elements is enough.

To reproduce enable all OSD elements (first row is sufficient), save and switch tab. Now disable mentioned elements (probably both).

Didn't check all the elements though. With current configurator it does not show this behavior so we can rule out firmware here.

Asizon, 2020-12-09:
Good catch @haslinghuis i have reproduced it, seems a clearly issue with these two Stick overlay elements

Asizon, 2020-12-09:
To reproduce no need to check all elements, just check Stick overlay and here is the error visible

haslinghuis, 2020-12-09:
Have the fix already but I have too many PR's open currently.

haslinghuis, 2020-12-09:
Found two possible ways to fix this:

1. First tried on line 223 putting the next 3 lines in an else clause which does the trick.
2. After further digging found we can also change line 2797 to:
```
const charCode = (element.sym !== undefined) ? element.sym : SYM.BLANK; 
```
Somehow unrelated but in the same category: Another intermittent problem found while quickly ticking the boxes is the error:
```
Firmware is transmitting a different number of statistics (null) to what the configurator is expecting (24)
decode @ osd.js:1974
```
Thoughts?

haslinghuis, 2020-12-11:
Somehow unrelated but in the same category and a reminder to look at: 

Another intermittent problem found while quickly ticking the boxes is the error:
```
Firmware is transmitting a different number of statistics (null) to what the configurator is expecting (24)
decode @ osd.js:1974
```

haslinghuis, 2020-12-12:
Also noticed line 648+659 a space in one case but not the other?

closed 2020-12-13 by mikeller

mikeller, 2020-12-13:
Not sure this should be closed, as not sure that #2332 really fixes this, so feel free to re-open if the problem still exists.

PR #2332 (2020-12-13): Fix drawStickOverlayPreview in OSD
Fixes: #2330