betaflight/betaflight-configurator #2339: Betaflight configurator becomes unresponsive when clicking tab after connect

svefro, 2020-12-14:
After connecting betalight to a flight controller everything seems to work until i click any of the tabs, the the whole app becomes unresponsive without any error messages.

I have tested 2 different flight controllers on this pc with the same result.

works fine on another pc.

I have tried uninstalling and reinstalling several times and deleted the betalight folder.

Firmware updating works and i can see the angle of the quad (live) after connecting until i click a tab.

haslinghuis, 2020-12-14:
Please try another good quality usb data cable. Please provide more information about version of the configurator and Operating System. Did it work before?

svefro, 2020-12-15:
i built from source and then it worked. tried nightly installer build and it worked also.
windows 10. i did not have it installed before

Seems like the issue was fixed, i was using version 10.7.0

-Svein

closed 2020-12-15 by svefro

Asizon, 2020-12-15:
@svefro then it is related to the graphic driver  issue in 10.7.0, the new dev 10.8.0 version solves it with Nw.js update.

svefro, 2020-12-15:
Sounds right. The copter 3d model on 10.7 was just a square, now it shows
the quad.