betaflight/betaflight-configurator #2340: Flashing in the android app??

butterSnake, 2020-12-14:
Is this possible as a feature ?

Asizon, 2020-12-14:
I think @WalcoFPV tried to implement it, but there are something dificult in this implementation. But it should be possible using a Cordova plugin for stm flashings.

mikeller, 2021-01-14:
Yeah, the problem is that there is no native driver for the STM DFU mode USB device in Android (as opposed to the ACM device that the USB port registers as when Betaflight is running).
So in order to make this work we'd have to implement and maintain an Android low level driver, which is considerably more involved than maintaining an android application.

UnklBilly1, 2024-03-09:
I'm just trying to get my drone to fly. I don't know what any of this means.

sugaarK, 2024-03-10:
> I'm just trying to get my drone to fly. I don't know what any of this means.

Use a computer not a fone

haslinghuis, 2024-05-06:
Not possible

closed 2024-05-06 by haslinghuis