betaflight/betaflight-configurator #2344: Ubuntu Configurator not requesting to run as sudo  version 10.7.0

KevinBeyel, 2020-12-16:
No idea why, I was going crazy trying to get it running. This may be because I'm running on strange hardware (surface pro 3); I'd judge low priority. Seems like a niche issue though

the FC (if it matters?) is a Crazybee F4 v3 pro

 OS: Ubuntu 20.10 groovy
 Kernel: x86_64 Linux 5.8.0-33-generic
 Uptime: 1h 9m
 Packages: 2452
 Shell: bash 5.0.17
 Resolution: 2160x1440
 DE: GNOME 3.38.1
 WM: Mutter
 WM Theme: Adwaita
 GTK Theme: Yaru [GTK2/3]
 Icon Theme: Yaru
 Font: Ubuntu 11
 Disk: 16G / 20G (82%)
 CPU: Intel Core i3-4020Y @ 4x 1.5GHz [53.0°C]
 GPU: Mesa DRI Intel(R) HD Graphics 4200 (HSW GT2)
 RAM: 2616MiB / 3837MiB

mikeller, 2020-12-16:
Not sure why you think that configurator should 'run as sudo'?

thewidgetsmith, 2021-01-16:
@KevinBeyel you should check the ownership of the device file under `/dev` and then make sure you are a member of that group. For example, for me my drone FC connects on `/dev/ttyACM0` which is owned by the `root` user and the `dialout` group. By adding my user to the `dialout` group, BetaFlight Configurator is completely happy running as a non-root process in Ubuntu 20.10.

closed 2021-02-25 by github-actions[bot]