betaflight/betaflight-configurator #2358: Resource command is not working

limonspb, 2020-12-28:
[image]

CLI `resource......`  command not working on the current master branch.

Steps to reproduce:
1. Open CLI
2. Type resource
3. Copy-paste any resource command from the output, press enter.

Desired behavior:
Resource command being sent, the resource being reassigned.

Actual behavior:
```
# null
###ERROR IN input: UNKNOWN COMMAND, TRY 'HELP'###
```
The issue appeared after this pull request is merged:
https://github.com/betaflight/betaflight-configurator/pull/2231
in particular file `CliAutoComplete.js`

McGiverGim, 2020-12-28:
Thanks for reporting! I will look into it tomorrow.

McGiverGim, 2020-12-29:
@limonspb I can't reproduce the problem. If I copy&paste the output of `resource` command in my case I don't have any error. It applies the resources without problem.

Maybe related to something more? Someone else can confirm the issue to see if we can find more common info?

limonspb, 2020-12-30:
@McGiverGim after a little bit more testing:
The bug happens only when this little hint is visible and I push enter
If I hide this little hint first, then the command goes throug.

[image]

closed 2020-12-31 by mikeller