betaflight/betaflight-configurator #2359: LED Tab is brocken

TheIsotopes, 2020-12-28:
since #2319 the LED tab is brocken ... @haslinghuis pls can you check whats wrong

After selecting the tab the configurator hangs, only a disconnect brings the configurator back to work again.

befor this pr: 
[image]

after this pr:
[image]

haslinghuis, 2020-12-29:
@TheIsotopes Are there any `console.log()` messages ? (F12) 
If it hangs there is probably a undefined value.

TheIsotopes, 2020-12-29:
@haslinghuis here the console output:

[image]

haslinghuis, 2020-12-29:
Moving line 71 to line 12 should resolve this. Are you able to check this change before I create a bugfix?

TheIsotopes, 2020-12-29:
ok, doesn't get stuck anymore ... but the functions and overlays of the LED's are not displayed (see first screenshot above)

[image]

haslinghuis, 2020-12-29:
Okay can reproduce. Initial it's working but after save and revisiting it does not display the functions / overlay.

haslinghuis, 2020-12-30:
@TheIsotopes - I just reverted the `led_strip.js` on `master` from another working tree and it behaves exactly the same as this issue linked PR. With the last changes after saving the LEDs the colors are back but I can't see any of the overlay functions I have set. But this is also what I see with previous version.
The bug seems to be an extra or missing starting dot in the jQuery selector element. Found some other issues but they already were there before my change so this wouldn't have caused impact. Please could you test my last change in the linked PR.

TheIsotopes, 2020-12-31:
You are right, it is displayed correctly but after changing and saving the wrong information is displayed.

haslinghuis, 2021-01-03:
@TheIsotopes - Could you please test again?

TheIsotopes, 2021-01-03:
@haslinghuis Now everything seems to work as usual again. 👍

PR #2362 (2021-01-05): Fix led_strip overlay
Fixes: #2359
closed 2021-01-05 by mikeller