betaflight/betaflight-configurator #2370: Sliders / Multipliers beyond 2.0x

harry1453, 2021-01-02:
I'm currently tuning my quad and I really like using the sliders, they really make it easier to tune.

However, my quad seems to want slightly lower than default I gain and much higher than default P/D gain. So my master multiplier is at 0.8x and my PD gain multiplier is at 2.0x. I would like to add more P and D gain, and I can't, as I have hit the limit. Obviously I could continue modifying individual values but sliders are much easier.

Please could you allow the user to (in expert mode only) manually input a multiplier (eg. type in 2.1x, 3x, etc to the left of each slider) to go beyond the limits of the slider?

harry1453, 2021-02-02:
@mikeller I came up with a basic design:

[image]

Would this be an acceptable contribution if I were to fully implement it?

haslinghuis, 2021-02-02:
Sliders are undergoing change towards implementation using firmware sliders for upcoming release. The legacy sliders will be frozen. Now there are 7 sliders.

harry1453, 2021-02-02:
@haslinghuis OK, so I presume that makes this issue irrelevant then? Or at least no longer relevant to this repo.

When you say sliders will be frozen, you mean code-frozen? or not movable? will we be able to move them if we continue to run BF4.2 and don't upgrade firmware?

haslinghuis, 2021-02-02:
Legacy sliders would be frozen. Maybe your proposal could be beneficial for the firmware sliders too.

harry1453, 2021-02-02:
Okay, thanks. Possibly but assuming that it is controllable via CLI a numeric value could be input that way. Min/max values will still be annoying though.

haslinghuis, 2021-02-02:
Look at PR 8118 in firmware for details

haslinghuis, 2021-02-02:
Start at PR 9119 in firmware for details

spatzengr, 2021-02-02:
I think this is a good add, but I also wanted to ask:
If you had an i-Term slider, would you be moving the Master down to 0.8 (which is what is limiting your range).
[image]

Note this many sliders would ONLY be visible when you toggle to "Adv." in the header.  When set to Basic (the default), you would ONLY see:
[image]

closed 2021-03-13 by github-actions[bot]