betaflight/betaflight-configurator #2372: Feature request

SPCORRIV, 2021-01-03:
VTX Power Mode

[image]

Would like to be able change VTX power from transmitter. This could be done via an Aux switch. I set it up in CLI but it does not seem to work, Could this be added in Modes panel, or add this feature in adjustments drop down. Either one would work, I guess whichever would be easier.

Thank You

Scott

mikeller, 2021-01-03:
This will need work in the firmware as well, since `vtxChannelActivationConditions` aren't currently available in MSP. In general, this is badly integrated into the rest of the firmware, and could probably do with some rework.

SPCORRIV, 2021-01-03:
I was able to get it to work, just a round about way, hoping this can be added in the future.

haslinghuis, 2024-05-06:
This is already implemented

closed 2024-05-06 by haslinghuis