betaflight/betaflight-configurator #2381: Betaflight-configurator app doesn't launch on Ubuntu 20.10

Emeric-dufour, 2021-01-12:
Hello,

I downloaded betaflight configurator release 10.7.0 on my ubuntu 20.10 OS but nothing happens when I launch the app after installation.
I used the file "betaflight-configurator_10.7.0_amd64.deb" for installation and there is no problem with installation.
Tried to change exec command to "sudo /opt..." in the betaflight-configurator.desktop file but still nothing happens after launch and sudo identification...

Is there a fix ?

mikeller, 2021-01-12:
@Emeric-dufour:

Delete the `/opt/betaflight/betaflight-configurator/`, `~/.config/betaflight-configurator`, and `~/.cache/betaflight-configurator/` directories and try again.

closed 2021-02-20 by github-actions[bot]

ClementYap, 2024-03-26:
I am also facing the same issue. How to resolve it?